getumbrel/umbrel-os #9: Generate custom torrc config

nolim1t, 2020-02-27:
Use the following
```tor --hash-password <password>```

But we might need to have a default for storing secrets. Maybe use the LND unlock password to do this right now?

mayankchhabra, 2020-05-30:
Let's use the dashboard's password (which is also lnd's password) (hashed and stored in `$HOME/db/user.json`) for this too! Consistency FTW!

nolim1t, 2020-05-31:
the only issue is this has to be generated before any user logs in.

nolim1t, 2020-05-31:
Im not going to bother with encrypting them.. they will never be exposed externally and the impact is extremely low. because they are not encrypted in the config files anyway

nolim1t, 2020-06-25:
Once https://github.com/getumbrel/umbrel-compose/pull/28 is released will add in the changes into rc.local or have it run configure-box.sh on first run

mayankchhabra, 2020-07-10:
Fixed.

closed 2020-07-10 by mayankchhabra