getumbrel/umbrel-os #22: Upgrade OS to 64-bit

mayankchhabra, 2020-05-28:
Raspberry Pi OS (formerly Raspbian) 64-bit [has finally been released](https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=275370) in beta! We should use it as our base image.

mayankchhabra, 2020-06-02:
Manually tested [umbrel-compose](https://github.com/getumbrel/umbrel-compose) on RPi 64-bit OS. Works great!!!

[image]

Waiting for pi-gen to [update their tool with 64-bit support](https://github.com/RPi-Distro/pi-gen/issues/413).

nolim1t, 2020-06-04:
just need their source

AaronDewes, 2020-06-29:
You can use this until they release their source code: https://github.com/RPi-Distro/pi-gen/pull/425

mayankchhabra, 2020-06-30:
Thanks, @AaronDewes. What is the difference between this and the official RPi 64-bit OS?

AaronDewes, 2020-06-30:
I don't know. I don't use the official Raspberry Pi OS, so I'm not sure what's different, but it should be mostly the same. This uses Debian as base (the official OS does the same), but all additional packages in Raspbian (except the ones which Raspberry Pi doesn't provide an ARM64 version yet).

AaronDewes, 2020-06-30:
I could submit a PR here if you want. Do you want to only have ARM64 or both?

mayankchhabra, 2020-06-30:
Thanks, @AaronDewes.

We'll test out your branch over the coming weeks to see if everything works OK and is compatible with our stack. Once everything is tested, we'll ditch the 32-bit OS completely as RPi 4 is our primary target device.

If you wanna help out with it, you're more than welcome!

nolim1t, 2020-07-01:
I think arm64 only is fine

AaronDewes, 2020-07-01:
They created their own ARM64 branch now.

nolim1t, 2020-07-03:
awesome. Feel free to send a PR if you want. Would love to test it.

The current project maintainers are stretched thin against many things.

AaronDewes, 2020-07-03:
I'll do that. Currently testing.

nolim1t, 2020-07-03:
Cool thanks

PR #43 (2020-07-28): Arm64
This adds ARM64 support. I have built it, but not tested yet. Fixes #22
closed 2020-07-28 by mayankchhabra