getumbrel/umbrel-os #23: Install fail2ban

mayankchhabra, 2020-05-28:
Install fail2ban to prevent SSH brute-force attacks. The default configuration should be fine as it is.

nolim1t, 2020-05-30:
Could be something for later. Right now theres not much reason for giving ssh access to the outside world when the dashboard does most things.

And anyone with ssh access will probably know what to do

mayankchhabra, 2020-05-30:
Agree, but since SSH is enabled by default and users already have access to it, it's better to at least limit the number of password retries. 

OR, we can disable SSH by default, and the option to enable can be inside the dashboard. I like this approach better, your thoughts?

nolim1t, 2020-05-31:
Will eventually disable all password access

closed 2020-12-01 by mayankchhabra