getumbrel/umbrel-os #30: Log to RAM to prevent SD degradation

mayankchhabra, 2020-06-01:
Not necessarily required right now, but definitely worth considering in the near future. https://github.com/azlux/log2ram

nolim1t, 2020-06-01:
good idea

mayankchhabra, 2020-06-09:
This a good thread for more ideas - https://www.thethingsnetwork.org/forum/t/how-can-microsd-flash-memory-wear-be-minimized/15896

nolim1t, 2020-06-09:
Will look at this after partitioner / swap

nolim1t, 2020-06-10:
Testing this out too

nolim1t, 2020-06-10:
Can't get it to work? Anyone else?

mayankchhabra, 2020-06-10:
What's the issue you're facing?

nolim1t, 2020-06-10:
doesnt run at all

nolim1t, 2020-06-25:
I'm more for making the SD card a read only filesystem, and putting everything (including moving docker images) onto the external media. SD card is a lot harder to replace

mayankchhabra, 2020-06-25:
If you've got it working like that, then sure let's try it. But if it's too much work, then just logging to RAM will save the SD from a ton of unnecessary hammering.

nolim1t, 2020-06-25:
probably be something as part of creating a partition.. stop docker, then move all the docker stuff over, and then start docker again.  Then make everything read only.

mayankchhabra, 2020-06-25:
Hmm feel free to give it shot if you want to. IMO we can stick with just offloading the logs to SD Card for the public beta, and create a RO filesystem on SD post that.

nolim1t, 2020-06-25:
Yes this is a nice to have..

mayankchhabra, 2020-12-01:
Replaced by https://github.com/getumbrel/umbrel/issues/206

closed 2020-12-01 by mayankchhabra