getumbrel/umbrel-os #31: Docker doesn't work in the latest image

mayankchhabra, 2020-06-02:
Docker daemon doesn't run by default. When trying to run it manually, it throws the following error:
[image]

Tested on:
RPi 4 / 4GB, running Umbrel OS v0.0.2.8

mayankchhabra, 2020-06-02:
Looks like it worked after a reboot? The Pi was probably rebooted twice or thrice before docker seemed to work: 

[image]

mayankchhabra, 2020-06-02:
Ok so after I did `docker-compose up`, it pulled the latest images but all the services ran into FS errors:

[image]

Did a reboot, and `docker-compose up` again and there were no more FS errors. This also happened when I ran `umbrel-compose` on the 64-bit RPi (first run caused FS errors which were fixed after a reboot), so this issue is probably not related to `umbrel-os`.

Looking good now:

[image]

nolim1t, 2020-06-02:
Looks like some hardware (media) issues? Im going to try it out soon

mayankchhabra, 2020-06-02:
Not entirely sure... but it looks like the compose systemd service isn't working too. The containers don't autostart on boot:

[image]

nolim1t, 2020-06-03:
Yes that is intentional not to autostart without the partitioning

nolim1t, 2020-06-03:
btw I also have no issues with the image also

meeDamian, 2020-06-03:
If you updated from very old images, you might need to `chown` data directories.

mayankchhabra, 2020-06-03:
It was a fresh install on a new SD, so no old data :|

nolim1t, 2020-06-04:
I can't reproduce this. I've already burned two images and installed on two different PIs

mayankchhabra, 2020-06-04:
I'll retry 3rd time, nice and slow

mayankchhabra, 2020-06-06:
Run 3 on Raspberry Pi 4 / 4GB running release v0.0.2.8

Errored out again:
[image]

`dockerd` logs (from `/var/log/daemon.log`):

```
May 31 15:56:31 umbrel fake-hwclock[129]: Sun 31 May 15:56:27 UTC 2020
May 31 15:56:31 umbrel systemd-fsck[142]: e2fsck 1.44.5 (15-Dec-2018)
May 31 15:56:31 umbrel systemd-fsck[142]: Superblock last write time is in the future.
May 31 15:56:31 umbrel systemd-fsck[142]: #011(by less than a day, probably due to the hardware clock being incorrectly set)
May 31 15:56:31 umbrel systemd-fsck[142]: rootfs: clean, 56619/137632 files, 443862/549888 blocks
May 31 15:56:31 umbrel systemd[1]: Started udev Coldplug all Devices.
May 31 15:56:31 umbrel systemd[1]: Starting Helper to synchronize boot up for ifupdown...
May 31 15:56:31 umbrel systemd[1]: Started Set the console keyboard layout.
May 31 15:56:31 umbrel systemd[1]: Started File System Check on Root Device.
May 31 15:56:31 umbrel systemd[1]: Starting Remount Root and Kernel File Systems...
May 31 15:56:31 umbrel systemd[1]: Started Helper to synchronize boot up for ifupdown.
May 31 15:56:31 umbrel systemd[1]: Started Remount Root and Kernel File Systems.
May 31 15:56:31 umbrel systemd[1]: Condition check resulted in Rebuild Hardware Database being skipped.
May 31 15:56:31 umbrel systemd[1]: Starting Create System Users...
May 31 15:56:31 umbrel systemd[1]: Starting Load/Save Random Seed...
May 31 15:56:31 umbrel systemd[1]: Starting Flush Journal to Persistent Storage...
May 31 15:56:31 umbrel systemd[1]: Started Load/Save Random Seed.
May 31 15:56:31 umbrel systemd-sysusers[152]: Creating group systemd-coredump with gid 996.
May 31 15:56:31 umbrel systemd-sysusers[152]: Creating user systemd-coredump (systemd Core Dumper) with uid 996 and gid 996.
May 31 15:56:31 umbrel systemd[1]: Started Flush Journal to Persistent Storage.
May 31 15:56:31 umbrel systemd[1]: Started Create System Users.
May 31 15:56:31 umbrel systemd[1]: Starting Create Static Device Nodes in /dev...
May 31 15:56:31 umbrel systemd[1]: Started Create Static Device Nodes in /dev.
May 31 15:56:31 umbrel systemd[1]: Reached target Local File Systems (Pre).
May 31 15:56:31 umbrel systemd[1]: Starting udev Kernel Device Manager...
May 31 15:56:31 umbrel systemd[1]: Started udev Kernel Device Manager.
May 31 15:56:31 umbrel systemd-udevd[160]: Using default interface naming scheme 'v240'.
May 31 15:56:31 umbrel systemd[1]: Found device /dev/serial1.
May 31 15:56:31 umbrel systemd-udevd[163]: Process '/usr/sbin/alsactl -E HOME=/run/alsa restore 0' failed with exit code 99.
May 31 15:56:31 umbrel systemd-udevd[168]: Using default interface naming scheme 'v240'.
May 31 15:56:31 umbrel systemd[1]: Found device /dev/disk/by-partuuid/7740b94a-01.
May 31 15:56:31 umbrel systemd[1]: Listening on Load/Save RF Kill Switch Status /dev/rfkill Watch.
May 31 15:56:31 umbrel systemd[1]: Condition check resulted in FUSE Control File System being skipped.
May 31 15:56:31 umbrel systemd[1]: Condition check resulted in Rebuild Hardware Database being skipped.
May 31 15:56:31 umbrel systemd[1]: Condition check resulted in Set Up Additional Binary Formats being skipped.
May 31 15:56:31 umbrel systemd[1]: Condition check resulted in Huge Pages File System being skipped.
May 31 15:56:31 umbrel systemd[1]: Starting File System Check on /dev/disk/by-partuuid/7740b94a-01...
May 31 15:56:31 umbrel systemd[1]: Starting Load/Save RF Kill Switch Status...
May 31 15:56:31 umbrel systemd[1]: Started Load/Save RF Kill Switch Status.
May 31 15:56:31 umbrel systemd-fsck[274]: fsck.fat 4.1 (2017-01-24)
May 31 15:56:31 umbrel systemd-fsck[274]: /dev/mmcblk0p1: 308 files, 106175/516190 clusters
May 31 15:56:31 umbrel systemd[1]: Started File System Check on /dev/disk/by-partuuid/7740b94a-01.
May 31 15:56:31 umbrel systemd[1]: Mounting /boot...
May 31 15:56:31 umbrel systemd[1]: Mounted /boot.
May 31 15:56:31 umbrel systemd[1]: Reached target Local File Systems.
May 31 15:56:31 umbrel systemd[1]: Starting Commit a transient machine-id on disk...
May 31 15:56:31 umbrel systemd[1]: Starting Preprocess NFS configuration...
May 31 15:56:31 umbrel systemd[1]: Starting Raise network interfaces...
May 31 15:56:31 umbrel systemd[1]: Starting Set console font and keymap...
May 31 15:56:31 umbrel systemd[1]: Starting Create Volatile Files and Directories...
May 31 15:56:31 umbrel systemd[1]: nfs-config.service: Succeeded.
May 31 15:56:31 umbrel systemd[1]: Started Preprocess NFS configuration.
May 31 15:56:31 umbrel systemd[1]: Condition check resulted in RPC security service for NFS server being skipped.
May 31 15:56:31 umbrel systemd[1]: Condition check resulted in RPC security service for NFS client and server being skipped.
May 31 15:56:31 umbrel systemd[1]: Reached target NFS client services.
May 31 15:56:31 umbrel systemd[1]: Reached target Remote File Systems (Pre).
May 31 15:56:31 umbrel systemd[1]: Reached target Remote File Systems.
May 31 15:56:31 umbrel systemd[1]: Started Set console font and keymap.
May 31 15:56:31 umbrel systemd[1]: Started Create Volatile Files and Directories.
May 31 15:56:31 umbrel systemd[1]: Starting Update UTMP about System Boot/Shutdown...
May 31 15:56:31 umbrel systemd[1]: Starting Network Time Synchronization...
May 31 15:56:31 umbrel systemd[1]: etc-machine\x2did.mount: Succeeded.
[... 393 lines cut ...]
Jun  6 04:01:12 umbrel systemd[1]: docker.service: Start request repeated too quickly.
[... cut ...]
Jun  6 04:02:21 umbrel systemd[1]: Started LSB: Resize the root filesystem to fill partition.
Jun  6 04:02:21 umbrel systemd[1]: systemd-update-utmp-runlevel.service: Succeeded.
Jun  6 04:02:21 umbrel systemd[1]: Started Update UTMP about System Runlevel Changes.
Jun  6 04:02:21 umbrel systemd[1]: man-db.service: Succeeded.
Jun  6 04:02:21 umbrel systemd[1]: Started Daily man-db regeneration.
Jun  6 04:02:21 umbrel systemd[1]: Startup finished in 1.050s (kernel) + 3min 54.356s (userspace) = 3min 55.407s.
```

PR #35 (2020-06-08): Use convenience script to install Docker
The error encountered in  #31 was due to a year old version of Docker being installed (`18.09.1`). It turns out that installing `docker.io` package isn't the supported method of installation anymore. 

> For Raspbian, installing using the repository is not yet supported. You must instead use the convenience script.
>
> *- [Docker's documentation](https://docs.docker.com/engine/install/debian/)*

Please test the [v0.0.2-build.10 image](https://github.com/getumbrel/umbrel-os/releases/tag/v0.0.2-build.10) on your RPi before merging.

Tested on RPi 4:

[ima [... cut]
closed 2020-06-08 by mayankchhabra