getumbrel/umbrel-os #32: Bundle Umbrel Compose's Docker Images

mayankchhabra, 2020-06-02:
**Issue**
Even though Umbrel OS ships pre-bundled with [Umbrel Compose](https://github.com/getumbrel/umbrel-compose), it has to pull all the docker images of services defined in `$HOME/docker-compose.yml` from docker hub on the first-boot. This adds a significant delay in the initial setup, thus having the user wait indefinitely to access the dashboard at http://umbrel.local without any method of keeping them updated with progress, ETA, or status of what's going on.

As we add more services to `umbrel-compose`, this will only keep on increasing the delay.

**Solution**
Add a step in stage 2 of image generation which pulls and bundles all the docker images of services defined in `$HOME/docker-compose.yml`. Thus, the user will have to no longer wait to access their dashboard at http://umbrel.local on the first boot.

Also, there is no benefit of pulling images on the first boot as we're not pulling the latest images anyway (since they are pulled in accordance with their version tags defined in `$HOME/docker-compose.yml`) so bundling them does no harm.

nolim1t, 2020-06-02:
yeah its a good one. But also a tough one to solve

mayankchhabra, 2020-06-02:
If we just add a new step to [stage 2](https://github.com/getumbrel/umbrel-os/tree/master/stage2) called `10-pull-images` and do `docker-compose pull` inside it, would that not be enough?

nolim1t, 2020-06-03:
Maybe possible

nolim1t, 2020-06-04:
If you know how to solve the following let me know.

[image]

nolim1t, 2020-06-04:
sounds plausible but I'll wait till the partitioning stuff is fully merged before I take this on to minimize potential conflicts

PR #42 (2020-07-06): Bundle Umbrel Docker Images
Closes #32
closed 2020-07-06 by mayankchhabra