getumbrel/umbrel-os #39: Enable firewall

mayankchhabra, 2020-06-27:
Install Uncomplicated Firewall  — `ufw` — and close all ports except 80 (for Nginx) and 22 (to SSH).

nolim1t, 2020-06-29:
I don't think this should be needed imediatly. The box would most likely sit behind a NATTED network and is unreachable except through explicitly defined ports over TOR.

mayankchhabra, 2020-06-29:
Yeah but an open port is always an attack surface, and even if the node isn't attacked, it can still reveal a lot of info. Plus this doesn't require a lot of effort to implement anyway, probably 10 minutes or so — just install the package and open port 80 and 22

lukechilds, 2020-07-24:
By default Docker edits iptables directly to setup port forwarding rules so it'll bypass any ufw rules we add.

There is a way to configure Docker play nice with custom iptables rules too but it's a little more involved.

However, is there really a benefit to this?

If we aren't running anything on the host apart from SSH, and everything inside Docker will be respecting the rules we setup in `docker-compose.yml`, what purpose is ufw serving?

ufw is just an easy to use frontend for iptables. And so are Docker port rules. It sounds bad to not explicitly use a firewall but `docker-compose.yml` is a firewall for everything inside Docker which is where we're running everything.

@meeDamian do you have an opinion on this?

meeDamian, 2020-07-24:
That behaviour can be disabled by adding `{"iptables": false}` to `/etc/docker/daemon.json`.

Otherwise, not sure if delegating this responsibility to compose is a good idea.  Might be, just the answer doesn't seem instantly obvious to me.

closed 2020-12-01 by mayankchhabra