getumbrel/umbrel-os #55: Extremely degraded performance on some USB 3.0 SSDs

mayankchhabra, 2020-07-22:
Many people, including myself, have had a painful experience of using an external SSD on a RPi 4 where the read/write speeds turn out to be less than 100 kB/s. 

Turns out that it's a known problem (and a pretty common one) where the Linux kernel has a non-exhaustive blacklist for devices known to be unreliable when using UAS [[full details here](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=245931)].

The fix is pretty straightforward though, just need to add a single line to `/boot/cmdline.txt`, so I think it's worthwhile to patch it for all RPi 4 builds (including those whose drives aren't blacklisted).

lukechilds, 2020-07-29:
Possibly related: https://github.com/getumbrel/umbrel-os/issues/64

nolim1t, 2020-07-29:
no its not. 

That issue is with RPI3.. on a RPI3 the dashboard and PI becomes almost unusable while bitcoind is syncing.

mayankchhabra, 2020-08-15:
Closed by #95 for two of the most popular USB-to-SATA chipsets. We can manually add other chipsets to the quirk if and when our users bump into this problem.

closed 2020-08-15 by mayankchhabra