getumbrel/umbrel-os #58: [Bug] 504 Gateway Time-Out Error shows plain HTML in Error Box

qnjctckzz8, 2020-07-23:
### Hello everyone!

_Sorry it took me a while, wanted to clarify and investigate the error before I submit an issue.
To pick up on the twitter conversation with @mayankchhabra_ 

### 504 Gateway Time-out Error shows plain HTML in Error Box

[image]

### Tasks
- [ ] Fix Error Message Display of Error Code "504".

PS: Should I just submit Issues for every finding I have in the future or should I make a spreadsheet and you pick whats important? Don't want to flood your issue list 😄

mayankchhabra, 2020-07-24:
Hey, thanks for reporting this issue @qnjctckzz8!

I can see there's something wrong with the Tor address too (on the Tor card, underneath "Remove Access", it says "false" instead of showing the hidden service URL).

Can you please share the following details?

- What device are you running Umbrel on?
- Are you using Umbrel OS or manual installation of Umbrel?
- Which browser are you using?

Re: opening issues, yes please feel free to flood us by opening issues for all of your findings. Don't worry, we'll close/transfer issues if they're not a priority or irrelevant.

qnjctckzz8, 2020-07-24:
Yes, I thought that would be a problem aswell - but tbh I couldn't replicate it after restarting the node and now it shows the correct URL.

- RaspberryPi 3 B+ & 500GB HDD
- Umbrel OS
- Screenshot via Brave, Looks the same via Firefox and Edge.

mayankchhabra, 2020-07-24:
Ah okay, thanks for the details, we'll try to figure out the reason why it didn't work the first time.

Meanwhile, I'll leave this issue open here for now, so if you re-encounter the bug please share the same here.

mayankchhabra, 2020-12-01:
Replaced by https://github.com/getumbrel/umbrel-dashboard/issues/222

closed 2020-12-01 by mayankchhabra