getumbrel/umbrel-os #63: Error - Not allowed by CORS

P-Verbrugge, 2020-07-24:
Tried Umbrel today. Flashed the image to a new 32GB SD card. The system booted, and I set my name and password. Then I got the 24 word list and wrote them all down. After the last word I tried to continue, but get the error "Not allowed by CORS". I am stuck at the last word screen and can not continue.

Also tried to redo the setup, and skipping the 24 words by selection the option "Note down later", but that gives me the same error.

mayankchhabra, 2020-07-24:
Hey thanks for trying out Umbrel and opening this issue, @P-Verbrugge!

Did you open the dashboard on `http://<ip-address-of-your-umbrel>`? If so, accessing the dashboard doesn't currently work with the IP address and we're [working on a fix for it](https://github.com/getumbrel/umbrel/issues/49). 

Meanwhile, you can try accessing the dashboard at `http://umbrel.local` instead and it should probably work.

However, if that's where you accessed it from and it didn't work, please let us know.

P-Verbrugge, 2020-07-24:
> Hey thanks for trying out Umbrel and opening this issue, @P-Verbrugge!
>
> Did you open the dashboard on `http://<ip-address-of-your-umbrel>`? If so, accessing the dashboard doesn't currently work with [...]

Ah, check. I did open the dashboard using the ip address. I got to the last screen now. It's says it syncing. :-)

Only noticed in the last screen, that the button "Go to dashboard" is not working (yet). When I access umbrel.local, I see the dashboard.

closed 2020-07-24 by P-Verbrugge

mayankchhabra, 2020-07-24:
Nice, good to know! 

Re: "Go to dashboard" not working, yeah we should add a loading spinner to it as it waits for LND to be ready.

kodxana, 2020-08-26:
Having that same error. Switched nginx container to forward port 2277 to 80 in container cause I have it occupied. Can't get past seed process.

mayankchhabra, 2020-09-01:
It usually happens when the hostname is not `umbrel`. But the dashboard should still be accessible (and working fine without a CORS issue) on the IP address.

Although we're working on a fix where any hostname is whitelisted by CORS dynamically at runtime.