getumbrel/umbrel-os #64: [Bug] 504 gateway timeout loop on initial site load & request flooding

qnjctckzz8, 2020-07-24:
### 504 gateway timeout loop on initial site load & request flooding
[image]

[image]

### setup
- Raspberry Pi 3 B+ & 500GB HDD
- UmbrelOS
- Brave Browser, Edge and Firefox
- **Node is syncing**

### problem information
- Initial site load took years, many requests timed out.
- After initial side load, the UI floods the API with requests while not receiving back any information, opening many duplicate connections.
- The Dashboard eventually loads, continuing to throw timeout errors and duplicate request warnings.

lukechilds, 2020-07-24:
>Problem seems (from my pov) to be that the system is overloaded during initial block sync and can't reliably answer api requests.

I think you mihgt be right, I think this may be a RAM issue.

Pi 3 only has 1GB of RAM and we're using:

```
dbcache=1000
maxmempool=512
```

In bitcoin.conf. So with 1GB of RAM we'll be constantly in swap.

Also as an aside:

>Raspberry Pi 3 B+ & 500GB HDD

The partitioner script checks for a drive with >500GB free space otherwise it sets up Umbrel with a pruned Bitcoin Core.
https://github.com/getumbrel/umbrel/blob/4263725016abf76b14b4789f193d145f67ae6db6/contrib/partitioner/partitioner.py#L132

So you won't be running a fully archival node, just a pruned node. And LND will be backed by Neutrino instead of Bitcoin Core. Are you aware of that?

(I know we need to make this clearer in the UI but we're not quite that polished yet)

qnjctckzz8, 2020-07-24:
**Re issue:**
That will probably explain it.
So SSH won't let me connect. I use the password from the readme (u: umbrel p: moneyprintergobrrr) but it says "Access denied." (connecting via putty). Different credentials?

**Re harddrive:** 
It's actually a 1TB harddrive, just thought the info 499+GB would be the important factor here (readme says >500GB is for full archival) but since the cutoff is not <500GB I will from now on be more precise ^^
I don't know how many people would use it but:
Being able to see your harddrive size in the UI & then deciding by myself (maybe with helper information) if I want to run pruned or fully archival would be the best UX for **me**.

lukechilds, 2020-07-24:
>So SSH won't let me connect. I use the password from the readme (u: umbrel p: moneyprintergobrrr) but it says "Access denied." (connecting via putty). Different credentials?

Can you try `umbr3lb0x`. The readme shows the latest changes but the password has been changed since the last release.

>Being able to see your harddrive size in the UI & then deciding by myself (maybe with helper information) if I want to run pruned or fully archival would be the best UX for me.

Completely agree with this. We should assume sensible default but users should be able to change the options which make sense for them. We have so many features we want to add and very limited resources so that's not super high priority right now but we've discussed similar things internally already.

qnjctckzz8, 2020-07-24:
**Re issue:**

**This is htop shortly after start of the pi**
[image]
**This is htop shortly after visiting umbrel.local**
_site was still very responsive_
[image]
**This is htop when it started throwing errors on mass**
_site was fairly unresponsive_
[image]

**Re features:**
No problem, from what I see you're a 3 man team so it's understandable. I would try to help but my brain is completely drained at the moment when I come from work, (mostly) frontend dev, since we have multiple high priority projects going on and I'm an apprentice trying to keep up with the pros 😄 - I hope doing a little QA / problem finding here and there will help! I plan to stay on it when I can.

lukechilds, 2020-07-25:
Hmmn, regarding those `htop` screenshots, `bitcoind` is sucking up about 75% of your RAM. That's quite a lot but it shouldn't be enough to make the web service unstable.

Could you try:

```
df- h /mnt/data
```

Just to check your SSD is being used.

Also might be worth to check `htop` again after it's been running of ran hour or two so `bitcoind`s dbcache has had time to fill up.

> I would try to help but my brain is completely drained at the moment when I come from work, (mostly) frontend dev, since we have multiple high priority projects going on and I'm an apprentice trying to keep up with the pros

Don't worry, I've been there. Don't burn out and look after your health!

qnjctckzz8, 2020-07-25:
>umbrel@umbrel:~ $ df -h /mnt/data
>Filesystem      Size  Used Avail Use% Mounted on
>/dev/sda1`       916G   14G  856G   2% /mnt/data

After waking up today (11h uptime) the webUI is completely not responding and stuck in 504 loop.

- The first 2 `ping` requests take 4 seconds each.
- The `status` requests end up timing out completely.
- `htop` doesn't look much different than in the last screenshot, 75% RAM by `bitcoind`

lukechilds, 2020-07-25:
It's unlikely to resolve the problem but can you try disabling cache in dev tools and hard refreshing? Just to rule cache issues out.

Also can you try restarting the services? There is a token that is created by the manager service and is used in the middleware and dashboard services. I'm wondering if something strange has happened there and manager has been restarted and has a new token but the others don't.

You can restart by either doing `docker-compose stop && docker-compose start` or do a full system reboot with `sudo reboot`.

qnjctckzz8, 2020-07-25:
Since you mentioned on twitter that it might be an issue I've disabled caching already, didn't make a difference - it's most definitely a performance problem.

`Rebooting` the device was what I did by myself, but did end up having the same issues only 1 1/2 hours after restarting.
I restarted via `docker-compose` now and the webUI is responsive again and getting the requests answered, but it most likely won't make a difference since the issue is getting worse the longer the up-time.

nolim1t, 2020-07-27:
Raspberry pi 3's tend to have a lot of performance issues while bitcoind is syncing post block 300k.

I do experience this myself on my pi 3s (but never an issue with my PI 4)..  Would be nice to know how to optimize better for PI 3 .

AaronDewes, 2020-07-27:
Can you run `htop` on tty while syncing? So we can see what resources it is using (I assume it's a ram issue and could be resolved with swap)

AaronDewes, 2020-07-27:
I think it's the USB writing speed on USB 2 what is causing the issue, but I'm not sure.

AaronDewes, 2020-07-27:
Please run `iotop` (in apt package `iotop` or `iotop-c`) while syncing to show disk usage.

lukechilds, 2020-07-27:
@qnjctckzz8 would you be able to try @AaronDewes suggestion and get back to us with the results?

qnjctckzz8, 2020-07-28:
Yea I wanted to yesterday but than the below happened and I wanted to give it some time:

So since I couldn't get it to work at all today, maybe I hit that ~300k and now it's completely done, I wanted to restart it one last time and, in case it wouldn't be reachable, just let it sync for a few weeks since I don't really need to have a 3rd working node right now.

But now I think I completely borked it.

[image]

Rebooting made it start the services again, but the status request of the webUI returns `operational: false` forever (I pre wrote the above 15 hours ago and it's still like this).

- bitcoind @ 72% memory usage according to htop
- iotop doesn't give any usefull information at this point because the only thing it does is read. 
There is not writing happening at this point and the blockchain size is stuck at 29GB.

[image]

[image]

nolim1t, 2020-07-28:
Do report back on if it even syncs past that .

qnjctckzz8, 2020-07-29:
[image]

So I left `iotop` open for a while to see whats happening ( >14 hours ) and it seems to be syncing, slowly but steady. 
Dashboard is reachable again aswell, but using it stops at the unlock request since that times out every time I send it.

nolim1t, 2020-07-29:
I've seen this issue before but only an issue on RPI3. It's basically a non issue on the 4.

Not too sure how to optimize bitcoind for the RPI 3 to give a better user experience

lukechilds, 2020-07-29:
@AaronDewes I'm not too familiar with the USB issue you mentioned, does the `iotop` output @qnjctckzz8 provided show you anything useful?

lukechilds, 2020-07-29:
Oops sorry, just seen your response @nolim1t. Is the USB issue @AaronDewes suspected or something unrelated?

lukechilds, 2020-07-29:
And @qnjctckzz8 do you by any chance have a Pi 4 you could test with the same SD/HDD on?

AaronDewes, 2020-07-29:
> I'm not too familiar with the USB issue you mentioned, does the iotop output @qnjctckzz8 provided show you anything useful?
Maybe, I would need a performance test.
 @qnjctckzz8 Please run on this command on the same Pi, with the same SSD, but with RPi OS, 2 or 3 times:

```
sudo hdparm -t /dev/sda
```

qnjctckzz8, 2020-07-29:
> dbcache=128
> maxmempool=128

[image]

Synced much more & much faster - so a small victory. But the dashboard still won't let me in.

> If that doesn't help, do you by any chance have a Pi 4 you could test with the same SD/HDD on?

Sadly none I can misuse at the moment, no.

> Maybe, I would need a performance test.
> @qnjctckzz8 Please run on this command on the same Pi, with the same SSD, but with RPi OS, 2 or 3 times:
>
> sudo hdparm -t /dev/sda

[image]

I can't do it with RPi OS right now because I don't have my cardreader but I did it on umbrelOS with the composer stoped, shouldn't make a difference or?
It's not the best performance considering it's halve of the theoretical max even of USB 2, but it's sort of the to be expected magnitude.

AaronDewes, 2020-07-29:
It doesn't make a difference. I assume RAM is an issue with this, because @lukechilds solution worked

qnjctckzz8, 2020-07-30:
So what I did for a test now was reducing the `dbcache` and `maxmempool` even further and I put `64` MB for both.
Until now, about 11 hours uptime, it looks good. The dashboard is reachable, unlockable & navigatable and it's syncing seemingly faster than before even with `128` MB.

Just don't even try changing the password. I don't know why this is a trigger, because it surely shouldn't be a resource intensive call, but it just will not be able to fullfill the request.

mayankchhabra, 2020-07-30:
> Just don't even try changing the password. I don't know why this is a trigger, because it surely shouldn't be a resource intensive call

Interesting. It could be resource-intensive on RPi 3, especially during the IBD, since it decrypts then re-encrypts stored secrets on your Umbrel, changes LND's password, and then restarts LND (which is usually what takes the longest time). 

Does it not work at all and if so, do you see any errors in the console?

Also, really appreciate your help in debugging these problems! Many thanks for doing this.

qnjctckzz8, 2020-07-30:
>Does it not work at all and if so, do you see any errors in the console?

The only console error I get is the gateway timeout.

>It could be resource-intensive on RPi 3, especially during the IBD, since it decrypts then re-encrypts stored secrets on your Umbrel, changes LND's password, and then restarts LND (which is usually what takes the longest time).

Hmm ok I didn't expect that. That will probably be the reason then.

Does it wait until all of these things have succeeded and THEN change the password and give a response to the API call?

Because if so... it most definitely explains why it takes more than 1 minute and therefore is a definite timeout:
When I do `docker-compose stop` - LND takes 1 minute+ ONLY to shut down.
No chance it does all of that within 1 minute 😆 

>Also, really appreciate your help in debugging these problems! Many thanks for doing this.

Hey no problem. I actually enjoy being on the other side of the table for a chance ^^

mayankchhabra, 2020-07-30:
Interesting, because we don't have a timeout defined for the API so even if it takes more than 1 minute it should be fine. 🤔

> Does it wait until all of these things have succeeded and THEN change the password and give a response to the API call?
And yeah, it does wait for the entire process to be finished before returning the response. 
 
Could you run `docker-compose logs -f manager`, and try changing the password on the dashboard to see if you could spot an anomaly in the logs somewhere?

qnjctckzz8, 2020-07-30:
Just FYI I'm testing with 32MB right now because I hoped having bitcoind and lnd not run on swapspace could lead to me being able to get a `changepassword` request through because of more free RAM.
This is the first time accumulative writes are higher than accumulative reads which I assumed would be the normal anyway.

>Could you run docker-compose logs -f manager, and try changing the password on the dashboard to see if you could spot an anomaly in the logs somewhere?

`manager              | ::ffff:10.11.0.2 - admin [Thu, 30 Jul 2020 17:10:39 GMT] "POST /v1/account/change-password HTTP/1.0" - - "http://umbrel.local/settings" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/84.0.4147.89 Safari/537.36"`
`manager              | umbrel-manager`
`manager              | Unable to change password`
`manager              | /v1/account/change-password`
`manager              | Error: Unable to change password`
`manager              |     at Object.changePassword (/usr/src/app/logic/auth.js:113:15)`
`manager              |     at async /usr/src/app/routes/v1/account.js:45:9`
`manager              |     at async /usr/src/app/utils/safeHandler.js:8:14`

That doesn't say much. Do you log more? Where?

>And yeah, it does wait for the entire process to be finished before returning the response.

If it waits to finish before giving a response, it would have changed the password before saying it changed the password. Right?
Therefore it wouldn't make a difference whether or not the request times out, the password would have changed.
Because the password does NOT change, we can assume that the request doesn't even get to the part where the magic would happen. Right?

AaronDewes, 2020-07-31:
> Interesting, because we don't have a timeout defined for the API so even if it takes more than 1 minute it should be fine. :thinking:

https://github.com/getumbrel/umbrel-manager/blob/5ea3852dbe916137d9daaa83356f978891375127/logic/auth.js#L59

I don't know if I understand this correctly, but I think it runs in a loop, always checks if any error happens, if it's restarting, it sleeps 1 sec, then tries again, but only if it's run less than 20 times, else it stops. This means a timeout of 20 secs.

mayankchhabra, 2020-08-02:
> If it waits to finish before giving a response, it would have changed the password before saying it changed the password. Right?

Yes, that's correct.

> it sleeps 1 sec, then tries again, but only if it's run less than 20 times, else it stops. This means a timeout of 20 secs.

Since it makes synchronous calls such as calling the password change API and disk read/writes, it will probably end up taking more than 1s per run and the sleep is just to delay the next run. So we can't infer the exact timeout.

qnjctckzz8, 2020-08-02:
Than I think we can close this issue here since https://github.com/getumbrel/umbrel/pull/73 is underway for the performance increase.

closed 2020-08-03 by mayankchhabra