getumbrel/umbrel-os #76: Hostname fails to resolve ~ every 24-48 hours

shawnyeager, 2020-07-31:
I'm seeing my ability to access the dashboard or ssh in via `umbrel.local` fail every 1-2 days. There's no doubt a more graceful way to resolve, but after `sudo reboot`, I'm back in action for another couple of days.

As a baseline, I am able to consistently ssh into various other devices on my network using their hostnames.

The details are over my head, so I don't know if this is something peculiar to my local network or not. FWIW, I'm using a single eero wifi router/AP running the latest firmware.

shawnyeager, 2020-07-31:
Update. Turns out that, at least with an eero, it's cycling hostnames on each reboot. No other device on my network does this.

[image]

nolim1t, 2020-08-01:
This sounds like a local issue.

Which is why we will be allowing for access via the IP address going forward because some devices do not resolve .local addresses at all (i.e. windows or android). 

You'll be able to see the IP address through your router or if you plug the box into an external display

nolim1t, 2020-08-01:
Closing as this is not an issue with umbrel-os and more of the router inadvertently changing settings without notifying (bad UX!)

mayankchhabra, 2020-08-01:
Hey, @shawnyeager keep us posted if you find a way to reserve a constant hostname (`umbrel.local`) for your Umbrel with Eero. This would be great to document!

shawnyeager, 2020-08-01:
FWIW, I have two other Pi's on the same network, neither of which have this problem. All, including the one running Umbrel, have IP reservations.

mayankchhabra, 2020-08-01:
Whoops, interesting. We'll leave this issue open here until we find a fix.

nolim1t, 2020-08-05:
Its not just eero router. I've seen this happen for me and a few people on telegram too

nolim1t, 2020-08-05:
This seems resolves itself when you restart avahi

nolim1t, 2020-08-05:
I might have a fix for this

nolim1t, 2020-08-05:
Testing out a fix for this now

nolim1t, 2020-08-06:
looks like what I did was working. I'll look at this issue

nolim1t, 2020-08-06:
I might have a fix to this, can you try https://github.com/getumbrel/umbrel-os/suites/1015613728/artifacts/13290451 and see if it works for you.

For all purposes this runs at the tip of the branch. I'll do my own testing too.

shawnyeager, 2020-08-06:
Thanks, @nolim1t. I've held off any other reinstalls until public beta, but will feedback then.

nolim1t, 2020-08-18:
still in the public beta btw ;)