getumbrel/umbrel-os #106: Required dependencies not installed after running `sudo ./build.sh`

53jk1, 2020-08-30:
Following the installation instructions on the Raspberry Pi, I get an error that dependencies are missing.
```
Required dependencies not installed

This can be resolved on Debian/Raspbian systems by installing:
quilt qemu-user-static debootstrap zerofree bsdtar bc
```
Perhaps we could add the installation of these dependencies directly to `build.sh` shell script.

AaronDewes, 2020-08-30:
This is the same in pi-gen: https://github.com/RPi-Distro/pi-gen. In my opinion, this should stay the same (but I'm not a maintainer).
Also, I'm not sure if you understand this repo correctly, it doesn't install Umbrel, it builds an OS image to flash. If you just want to install Umbrel, you can follow this guide: https://github.com/getumbrel/umbrel#installation-requirements, but if you follow this guide, Umbrel will not use an external drive.

You can "convert" the Raspberry Pi OS to a "real" Umbrel OS by running this (I haven't tested this, If you want to try it, it might break your system, delete files, try it at your own risk, but you can tell me if an error occurs):

```Bash
sudo apt install git fswatch jq python3-qrcode python3-pip -y
sudo curl -fsSL https://get.docker.com | sh
sudo usermod -a -G docker $FIRST_USER_NAME
sudo pip3 install docker-compose
sudo sed -i "s/#allow-interfaces=eth0/allow-interfaces=eth0,wlan0/g;" "/etc/avahi/avahi-daemon.conf";
mkdir ~/umbrel
cd ~/umbrel
curl -L https://github.com/getumbrel/umbrel/archive/v0.2.5.tar.gz | tar -xz --strip-components=1
cd scripts/umbrel-os/services
UMBREL_SYSTEMD_SERVICES=$(ls *.service)
for service in $UMBREL_SYSTEMD_SERVICES; do
    sed -i -e "s/\/home\/umbrel/\/home\/pi/g" "${service}"
    sudo install -m 644 "${service}"   "/etc/systemd/system/${service}"
    sudo systemctl enable "${service}"
done
sed -i -e "s/\/home\/umbrel/\/home\/pi/g" "~/umbrel/scripts/umbrel-os/umbrel-details"
cd ~/umbrel
IMAGES=$(grep '^\s*image' docker-compose.yml | sed 's/image://' | sed 's/\"//g' | sed '/^$/d;s/[[:blank:]]//g' | sort | uniq)

while IFS= read -r image; do
    sudo docker pull --platform=linux/arm64 $image
done <<< "$IMAGES"
sudo bash -c 'echo "UMBREL_OS=v0.2.5" >> /etc/default/umbrel'
```

53jk1, 2020-08-30:
`sed -i -e "s/\/home\/umbrel/\/home\/pi/g" "~/umbrel/scripts/umbrel-os/umbrel-details"`

At this point, the terminal tells me that it cannot read `~/umbrel/scripts/umbrel-os/umbrel-details`. There is no such file or directory. However, there is such a file. A bit strange.

mayankchhabra, 2020-08-30:
@53jk1 looks like you're trying to build the OS image from source. Is that on purpose? If you just want to run Umbrel on a Raspberry Pi you can straightaway download the latest OS image from here: https://github.com/getumbrel/umbrel-os/releases/latest

53jk1, 2020-08-30:
Yes, I tried to build an OS image from source. I wanted to experiment a bit but it failed. It remains for me to download the latest OS image.

mayankchhabra, 2020-08-30:
Ah okay, in that case, I agree with @AaronDewes that we should stick to the [pi-gen](https://github.com/RPi-Distro/pi-gen) repo's `build` script.

You may want to check out our GitHub Action that automatically builds OS image on an Ubuntu 18.04 machine. It has all the steps: https://github.com/getumbrel/umbrel-os/blob/master/.github/workflows/on-push.yml

closed 2020-09-09 by 53jk1