getumbrel/umbrel-os #109: Failed building from source - Ubuntu 20.04

billygarrison, 2020-08-30:
When building the image from source, an error message is displayed with the dependencies for Debian / Raspibian. I think those dependencies will also work for Ubuntu 18.04 but it isn't clear - it could be helpful to list the dependencies in the README.

Either way, on Ubuntu 20.04 (from Windows Subsystem for Linux), those dependencies don't work as there is no bsdtar binary in apt. I had to install libarchive-tools instead for it to start the build process. 

The build failed with the following, not sure if it is because of using libarchive-tools (which I think includes the bsdtar binary):
```
I: Extracting liblzma5...
I: Extracting zlib1g...
I: Running command: chroot /home/billy/src/umbrel-os/work/2020-08-30-umbrel-os/stage0/rootfs /debootstrap/debootstrap --second-stage
chroot: failed to run command ‘/debootstrap/debootstrap’: Exec format error
rmdir: failed to remove '/home/billy/src/umbrel-os/work/2020-08-30-umbrel-os/stage0/rootfs/debootstrap': Directory not empty
```

AaronDewes, 2020-08-31:
> not sure if it is because of using libarchive-tools
No, it isn't. 
Please try PR #110, it should help.

> When building the image from source, an error message is displayed with the dependencies for Debian / Raspibian. I think those dependencies will also work for Ubuntu 18.04 but it isn't clear - it could be helpful to list the dependencies in the README.

Ubuntu is Debian based, but there may be some differences in package names.

mayankchhabra, 2020-09-01:
> it could be helpful to list the dependencies in the README

I agree, we should definitely add them to the README.

> there is no bsdtar binary in apt

I'm not sure on this one. If you checkout our GH Action that automatically builds the Umbrel OS image on a push, [it does install `bsdtar` using apt](https://github.com/getumbrel/umbrel-os/blob/master/.github/workflows/on-push.yml#L17), and it's building on Ubuntu 18.04.

billygarrison, 2020-09-01:
> Please try PR #110, it should help
Thanks @AaronDewes I will try that.

> I'm not sure on this one. If you checkout our GH Action that automatically builds the Umbrel OS image on a push, it does install bsdtar using apt, and it's building on Ubuntu 18.04.
@mayankchhabra sorry my issue wasn't more clear. The difference between my setup and the GH Action is that mine uses Ubuntu 20.04 on WSL (Windows Subsystem for Linux), vs (Ubuntu 18.04 for GH action). Google tells me that bsdtar is no longer in apt for 20.04 (but it is for 18.04).

billygarrison, 2020-09-01:
I'm still getting the error using PR #110 

I tried it on Ubuntu 18.04 (using Windows Subsystem for Linux) and get a slightly different error message:

```
I: Extracting liblzma5...
I: Extracting zlib1g...
I: Running command: chroot /home/billy/src/umbrel-os/work/2020-09-01-umbrel-os/stage0/rootfs /debootstrap/debootstrap --second-stage
prepare_binprm: Bad file descriptor
rmdir: failed to remove '/home/billy/src/umbrel-os/work/2020-09-01-umbrel-os/stage0/rootfs/debootstrap': Directory not empty
```

hugoender, 2022-04-21:
I'm having this exact issue. I see when using Ubuntu WSL as well as when using a Raspberry Pi. Is there a fix for this? I followed all the PR's mentioned but I'm not seeing a fix or workaround.

hugoender, 2022-04-24:
> I'm having this exact issue. I see when using Ubuntu WSL as well as when using a Raspberry Pi. Is there a fix for this? I followed all the PR's mentioned but I'm not seeing a fix or workaround.

Nevermind. I got it working. Not exactly sure what fixed it since I changed a few things at once. Instead of using the Raspberry Pi OS Lite, I installed the full version and then I also installed Docker using the script they provide on their site as opposed to using apt which is what I used the first time.