getumbrel/umbrel-os #115: bitcoin container stopped working / corrupted on rpi 3

qnjctckzz8, 2020-09-03:
Hey - it's me again.
After we discussed the password change issue I just left it running for a few weeks (not even trying to connect to the dashboard).
Yesterday I tried connecting to the dashboard, but it returned `operational: false, unlocked: false` for more than an hour - which is when I decided to try the old `docker-compose stop - docker-compose start` technique. Unsuccesfully.
It seems like something went wrong along the way with the bitcoin container (see screenshots below).

[image]
[image]
[image]
[image]

AaronDewes, 2020-09-03:
This can be resolved by deleting the bitcoin database, which I think is stored on the external drive. I don't know why this happens.

nolim1t, 2020-09-04:
Looks like drive corruption. 

What sort of power supply do you use? (Rated Output).

Also what sort of power situation do you have *any* sort of power interruption which is known to cause issues either now or later. And corruption type issues are irreversible too.

qnjctckzz8, 2020-09-04:
> This can be resolved by deleting the bitcoin database, which I think is stored on the external drive.
Yes, that's what I did afterwards but I was more interested in the cause.

>What sort of power supply do you use? (Rated Output).
I don't know exactly but it's the official raspberry one - 5.1V 13W 2500mA?

So I was on vacation which is why I can't say for sure that the Pi3 didn't loose power in the meantime.
However, since my other Pi4, which also runs bitcoind & lnd and runs on the same power, hasn't had an issue, I expect there was no power outage.
If it's drive corruption, what may have happened is that the USB Hub or the Harddrive malfunctioned?

Either way it then most likely is a hardware error.

Timbo925, 2020-11-23:
> This can be resolved by deleting the bitcoin database, which I think is stored on the external drive. I don't know why this happens.

I was wondering what you mean by deleting the bitcoin database? Is this the full bitcoin folder under `/home/umbrel/umbrel/bitcoin` ? 

I hava a similar problem but can't seem to find the correct folder to delete.

mayankchhabra, 2020-11-23:
@Timbo925 you can do

```
# Stop Umbrel
sudo systemctl stop umbrel-startup

# Delete everything in Bitcoin Core's directory except the configuration file
rm -rf /home/umbrel/umbrel/bitcoin/!(bitcoin.conf)

# Remove IBD status file
rm /home/umbrel/umbrel/statuses/node-status-bitcoind-ready

# Switch LND to neutrino
sed -i "s/bitcoin.node=bitcoind/bitcoin.node=neutrino/g;" "/home/umbrel/umbrel/lnd/lnd.conf"

# Start Umbrel
sudo systemctl start umbrel-startup
```

Bitcoin Core will now sync from scratch and LND will automatically use it as the back-end once the sync is complete.

Timbo925, 2020-11-28:
@mayankchhabra Seemed to be a HDD problem in the end I think. I did a full fresh install and recovered with the seed.

After full sync none of the balance of the bitcoin or lightning node is present yet, are there other steps to take?

mayankchhabra, 2020-11-30:
@Timbo925 did you have lightning channels that were open? If so, do you also have the channel backup file with you?

Can you also please check if LND is fully synced:

```
cd ~/umbrel && docker-compose exec lnd lncli getinfo | grep synced
```

If either `synced_to_chain` or `synced_to_graph` is `false`, I suggest waiting a little more. Sometimes LND can be painfully slow to sync the on-chain (Bitcoin wallet) transactions.

Timbo925, 2020-11-30:
@mayankchhabra Both synced values true. (I do have the backup files to restore) 

I do think my deposits on-chain happened before the release which switched to segwit native addresses (bc1...), maybe this has something to do with the fact that the sync isn't picking the transaction?

mayankchhabra, 2020-11-30:
I don't think switch to native-segwit should make a difference since it's a client-side switch.

Can you please DM me on Telegram so I can help you debug the issue more efficiently: https://t.me/mayankchhabra

mayankchhabra, 2020-11-30:
@Timbo925 was able to recover his on-chain and off-chain funds. Closing this now as the original issue is regarding Pi 3, which we no longer support. Thanks @qnjctckzz8 for helping us test out stuff early on! :)

closed 2020-11-30 by mayankchhabra