getumbrel/umbrel-os #124: Add bash aliases for commonly-expected executables running inside Docker containers

vindard, 2020-09-17:
A useful abstraction for RPi installs could be to **_bash alias_** common-expected commands like `lncli` to the location where they're currently accessible (`docker exec -it lnd lncli`). Executables people usually try to get at are:
- `lncli`
- `bitcoin-cli`
- `lnd`
- `bitcoind`

[image]

nolim1t, 2020-09-17:
maybe put them in `/usr/local/bin` ?

vindard, 2020-09-20:
> maybe put them in `/usr/local/bin` ?

It looks like @lukechilds already did this in [an older PR](https://github.com/getumbrel/umbrel/pull/161) (:raised_hands:) and it's just to export that `PATH` variable automatically when umbrel-os is being installed if we wanted.

@mayankchhabra was suggesting to add it to [this file](https://github.com/getumbrel/umbrel-os/blob/master/stage2/04-set-umbrel-os-version/files/umbrel) in the repo I think. Would it be as simple as adding a new `PATH="$PATH:$HOME/umbrel/bin"` line there, and is that something that we'd want to do? I'm thinking if someone has found their way through SSH to the terminal already then they're probably fairly technical and it's ok for them to have easier access by default to those binaries :thinking:

AaronDewes, 2020-09-20:
> Would it be as simple as adding a new `PATH="$PATH:$HOME/umbrel/bin"` line there

Yes, /etc/profile runs `source /etc/default/umbrel`, and if this is added/etc/default/umbrel, it should be directly accessible.

> and is that something that we'd want to do?

@mayankchhabra and @lukechilds need to decide that.

mayankchhabra, 2020-12-02:
Closing this since we decided to not add the path to discourage manual CLI use (https://github.com/getumbrel/umbrel/pull/161#issuecomment-676102234) as the binary proxies were added primarily for debugging.

closed 2020-12-02 by mayankchhabra