getumbrel/umbrel-os #126: Login prompt on boot

AaronDewes, 2020-09-20:
Two users reported that Umbrel OS got stuck on login prompt in Telegram. I'm not sure why this happens, but this should not happen. After a manual start of Umbrel, Umbrel was working as normal.

AaronDewes, 2020-10-25:
Both cases were probably user errors, closing this issue now

closed 2020-10-25 by AaronDewes

lukechilds, 2020-10-25:
Just FYI, it's probably not a good idea to get users to manually start Umbrel. We have some scripts that will stop Umbrel if it detects unreliable external storage.

We don't wan't Umbrel to continue to run if the storage device is unreliable since it could cause serious problems.

The best thing to do in that scenario is to run `journalctl -u umbrel-external-storage` and see if it's logged an error about why it stopped Umbrel.