getumbrel/umbrel-os #129: Support for RockPro64

thunderbiscuit, 2020-09-28:
Hey guys!

Absolutely love the Umbrel project. I'm looking to get my own Umbrel working with testnet on a RockPro64 from [PINE64](https://www.pine64.org/). I'm not super familiar with the differences between the OS needed for Raspberry Pis and RockPros, but I just wanted to let you know that I'd be down for helping with the testing if that's something you're looking at supporting in the future!

Let me know if you have time/ideas as to where to start for that. Should I just try the OS as-is and report back?

AaronDewes, 2020-09-29:
I'm not sure what files are normally on the boot partition of a RockPro. Can you list them?

nolim1t, 2020-09-29:
you should be able to run it on rockpro already, as its from pine64. I have no issues running this on pine64 hardware as I use a Pinebook already.

Simply use the https://github.com/getumbrel/umbrel project instead and follow the instructions from there.

As for a complete solution, you'd want to take a look at the [pmbootstrap](https://gitlab.com/postmarketOS/pmbootstrap) and [pmaports](https://gitlab.com/postmarketOS/pmaports) project

thunderbiscuit, 2020-09-29:
Thanks guys!

@AaronDewes I'm not exactly sure what you mean. The files on the BIOS somehow? If instead you mean somewhere in the kernel or distro, the RockPro doesn't have an "official" OS like the Pi does. Not sure if that helps... let me know if I can help you further.

@nolim1t yeah totally I might just grab like the Manjaro ARM distro and do it using the instructions provided on the umbrel repo. But I also want to throw my support behind having the 1-click install that `umbrel-os` offers working on the RockPro if that's in the cards. The myNode project for example installs on a few different single board computers off the bat, which is a nice feature.

nolim1t, 2020-09-29:
I think building an alpine apk (postmarket os) package might be a way too. Then can use pmbootstrap to install

AaronDewes, 2020-09-29:
I've send you a DM on Telegram, I think I know how to get it to work.

thunderbiscuit, 2020-11-13:
I managed to get all of this working; thanks to everyone who helped me out! [Blog post is here](https://thunderbiscuit.com/umbrel-rockpro/).

I think this issue is ready to be closed.

mayankchhabra, 2020-11-30:
Great work, @thunderbiscuit! Thanks for penning down the blog post.

closed 2020-11-30 by mayankchhabra

fueleh, 2021-09-24:
Got this working on my pineA64 with @thunderbiscuit blog post thanks!