getumbrel/umbrel-os #141: Wifi instead of Ethernet Cable?

michaelWuensch, 2020-10-23:
Hi there,
I am currently running a raspiblitz with wifi and would like to switch to Umbrel to try it out. But before I kill my current node, I 'd like to know if it supports wifi. If yes, are there any insturctions on how to do the installation with wifi?
If no, I'd like to see that feature added.

Thanks for the great work you are doing!

lukechilds, 2020-10-23:
You can indeed use WiFi with Umbrel OS.

After flashing the SD card, remove it and then plug it back in to your computer. Should see it show up in your file manager with the name "boot".

You need to create a file called `wpa_supplicant.conf` with the following contents:

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=<Insert 2 letter ISO 3166-1 country code here>

network={
 ssid="<Name of your wireless LAN>"
 psk="<Password for your wireless LAN>"
}
```

Then unmount the SD card, put it in your Pi, and it should automatically conned to your WiFi when it boots.

Let us know how you get on!

AaronDewes, 2020-10-23:
You need to manually put a file onto the boot partition called "wpa_supplicant.conf" with an adjusted version of this as content:

```
country=us
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
 scan_ssid=1
 ssid="MyNetworkSSID"
 psk="Pa55w0rd1234"
}
```
You can remove the scan_ssid if your network SSID isn't hidden.

AaronDewes, 2020-10-23:
You can basically use the file from RaspiBlitz, they recommend doing it the same way.

michaelWuensch, 2020-10-23:
Awesome, thanks for the info!

closed 2020-11-05 by mayankchhabra

NayrNotrub, 2021-08-31:
> You can indeed use WiFi with Umbrel OS.
> 
> After flashing the SD card, remove it and then plug it back in to your computer. You should see it show up in your file manager with the name "boot".
> 
> You need to create a file called `wpa_supplicant.conf` with the following contents:
> 
> ```
> ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
> update_config=1
> country=<Insert 2 letter ISO 3166-1 country code here>
> 
> network={
>  ssid="<Name of your wireless LAN>"
>  psk="<Password for your wireless LAN>"
> }
> ```
> 
> Then unmount the SD card, put it in your Pi, and it should automatically conned to your WiFi when it boots.
> 
> Let us know how you get on!

I followed all of these steps and I cannot get my Node to work on WIFI. Any assistance is greatly appreciated:

I am running a Rasp pi4 with the latest Umbrel OS. I used TextEdit (this could be the problem but I don't know what I am doing)to creat the file above, copied into the Boot SD and it still doesn't work. Please let me know what other information you may need to help troubleshoot. Again, thanks for any assistance, its greatly appreciated.

alexzaitsev, 2022-03-22:
How to connect to WiFi with [Captive portals](https://openwrt.org/docs/guide-user/services/captive-portal/start)?

robiwalker, 2022-09-11:
Will I be able to do this retrospectively? i.e take the SD card out of a Pi that's already set up with umbrel and add the file afterwards..?

TheSCQuant, 2022-09-16:
What are the contents of the file if the wifi network requires a username and password?