getumbrel/umbrel-os #142: HDD failure. How to recover?

BTCMachineElf, 2020-10-27:
Title pretty much says it all. I have a funded node w/funded channel and the HDD died. Deader than dead. I will replace it in a few days, but it seems the recovery functionality hasn't been rolled out yet, so how can I recover? (i have my channel backup)

BTCMachineElf, 2020-10-27:
There's only one channel, and I can get the remote node operator to force close it if need be. But if theres a way to recover its preferable.

lukechilds, 2020-10-30:
Apologies for the delayed reply, yes, you can recover.

All you need is your seed phrase and you can recover from the Umbrel cloud backups. It's not implemented in the UI yet but we can talk you through the process.

BTCMachineElf, 2020-10-30:
We've spoken on telegram, you'll sort me out there. Thanks!

closed 2020-12-01 by mayankchhabra