getumbrel/umbrel-os #153: New install on Pi 4b 8gb, exit code 137 all docker containers

adamcollier1, 2020-12-19:
Hi

Installed Umbrel OS on a new Pi 4b, 8gb ram

During the process, all the docker containers just exit with the code (137)

`umbrel@umbrel:~ $ docker ps -a
CONTAINER ID        IMAGE                                COMMAND                  CREATED             STATUS                        PORTS               NAMES
3d9e2911a73e        getumbrel/middleware:v0.1.7          "docker-entrypoint.s…"   2 hours ago         Exited (137) 19 minutes ago                       middleware
4eeafda273b6        getumbrel/manager:v0.2.7             "docker-entrypoint.s…"   2 hours ago         Exited (137) 19 minutes ago                       manager
bd0811bae0ab        getumbrel/neutrino-switcher:v1.0.3   "/usr/local/bin/swit…"   2 hours ago         Exited (137) 19 minutes ago                       neutrino-switcher
fd50219b4433        getumbrel/electrs:v0.8.6             "electrs"                2 hours ago         Exited (137) 19 minutes ago                       electrs
25be5fff7417        nginx:1.17.8                         "nginx -g 'daemon of…"   2 hours ago         Exited (137) 19 minutes ago                       nginx
f53f150aea1b        lncm/bitcoind:v0.20.1                "bitcoind -zmqpubraw…"   2 hours ago         Exited (137) 19 minutes ago                       bitcoin
f237db53c50b        lncm/lnd:v0.11.1                     "/usr/local/bin/lnd"     2 hours ago         Exited (137) 19 minutes ago                       lnd
0976542770fd        getumbrel/tor:v0.4.1.9               "tor"                    2 hours ago         Exited (137) 19 minutes ago                       tor
a9952d75b054        getumbrel/dashboard:v0.3.13          "docker-entrypoint.s…"   2 hours ago         Exited (137) 19 minutes ago                       dashboard
fb6d33c9470e        getumbrel/frontail:v4.9.1            "/frontail/docker-en…"   2 hours ago         Exited (137) 19 minutes ago                       frontail`

1TB SSD attached

Rebooting and it continues to sync, let me know if theres anything else I can provide but from a little bit of searching online it seems it's relating to docker running out of memory

AaronDewes, 2020-12-19:
Please run this:

```
wget https://raw.githubusercontent.com/getumbrel/umbrel/4603b0357ae5f118847787cd8f4032b99e7947da/scripts/debug  -O ~/umbrel/scripts/debug && chmod +x ~/umbrel/scripts/debug && ~/umbrel/scripts/debug --upload
```

adamcollier1, 2020-12-19:
Managed to catch the crash this time tailing the logs in the Umbrel web gui

`Dec 19 13:19:56 umbrel bitcoin[767]: Error: A fatal internal error occurred, see debug.log for details
Dec 19 13:19:56 umbrel bitcoin[767]: 2020-12-19T13:19:56Z Unable to open file /data/.bitcoin/blocks/blk00000.dat
Dec 19 13:19:56 umbrel bitcoin[767]: 2020-12-19T13:19:56Z ERROR: ReadBlockFromDisk: OpenBlockFile failed for FlatFilePos(nFile=0, nPos=33335304)
Dec 19 13:19:56 umbrel bitcoin[767]: 2020-12-19T13:19:56Z *** Failed to read block
Dec 19 13:19:56 umbrel bitcoin[767]: 2020-12-19T13:19:56Z Error: A fatal internal error occurred, see debug.log for details
Dec 19 13:19:56 umbrel bitcoin[767]: 2020-12-19T13:19:56Z ERROR: ProcessNewBlock: ActivateBestChain failed (Failed to read block)
Dec 19 13:19:56 umbrel bitcoin[767]: 
Dec 19 13:19:56 umbrel bitcoin[767]: 
Dec 19 13:19:56 umbrel bitcoin[767]: ************************
Dec 19 13:19:56 umbrel bitcoin[767]: EXCEPTION: 15dbwrapper_error       
Dec 19 13:19:56 umbrel bitcoin[767]: Fatal LevelDB error: IO error: /data/.bitcoin/indexes/txindex/000375.log: Read-only file system       
Dec 19 13:19:56 umbrel bitcoin[767]: bitcoin in scheduler       
Dec 19 13:19:56 umbrel bitcoin[767]: 
Dec 19 13:19:56 umbrel bitcoin[767]: terminate called after throwing an instance of 'dbwrapper_error'
Dec 19 13:19:56 umbrel bitcoin[767]:   what():  Fatal LevelDB error: IO error: /data/.bitcoin/indexes/txindex/000375.log: Read-only file system
Dec 19 13:19:56 umbrel bitcoin[767]: 2020-12-19T13:19:56Z Fatal LevelDB error: IO error: /data/.bitcoin/indexes/txindex/000375.log: Read-only file system
Dec 19 13:19:56 umbrel bitcoin[767]: 2020-12-19T13:19:56Z You can use -debug=leveldb to get more complete diagnostic messages
Dec 19 13:19:56 umbrel bitcoin[767]: 2020-12-19T13:19:56Z 
Dec 19 13:19:56 umbrel bitcoin[767]: 
Dec 19 13:19:56 umbrel bitcoin[767]: ************************
Dec 19 13:19:56 umbrel bitcoin[767]: EXCEPTION: 15dbwrapper_error       
Dec 19 13:19:56 umbrel bitcoin[767]: Fatal LevelDB error: IO error: /data/.bitcoin/indexes/txindex/000375.log: Read-only file system       
Dec 19 13:19:56 umbrel bitcoin[767]: bitcoin in scheduler       
Dec 19 13:19:56 umbrel bitcoin[767]: 
Dec 19 13:19:57 umbrel bitcoin[767]: Error: Cannot write to data directory '/data/.bitcoin'; check permissions.`

I will run the debug script linked shortly

AaronDewes, 2020-12-19:
Can you contact me on Telegram/Discord? That would make debugging easier.

adamcollier1, 2020-12-19:
Sure, what's your discord? I can't see it on your github profile

AaronDewes, 2020-12-19:
Here is an Umbrel discord server: https://discord.gg/EPBMn6T3

The #telegram channel is linked to The Umbrel telegram group, so others can help you too!

hitchcott, 2020-12-21:
I've also been running into this issue with `umbrel-os-v0.2.15` and Pi 4B 4GB.

Seems like the IO errors might be something to do with power supply. I have ordered an official one to see if this sorts it out.

@adamcollier1 are you using an official power supply?

```bash
umbrel@umbrel:~ $ docker ps -a
CONTAINER ID        IMAGE                                COMMAND                  CREATED             STATUS                       PORTS               NAMES
1b23bf631313        getumbrel/neutrino-switcher:v1.0.3   "/usr/local/bin/swit…"   7 minutes ago       Exited (137) 2 minutes ago                       neutrino-switcher
b29f319b1821        getumbrel/electrs:v0.8.6             "electrs"                7 minutes ago       Exited (137) 2 minutes ago                       electrs
c6f4c2e46748        lncm/bitcoind:v0.20.1                "bitcoind -zmqpubraw…"   7 minutes ago       Exited (1) 2 minutes ago                         bitcoin
4ba55bf48ae1        lncm/lnd:v0.11.1                     "/usr/local/bin/lnd"     7 minutes ago       Exited (137) 2 minutes ago                       lnd
07aadb3d3394        nginx:1.17.8                         "nginx -g 'daemon of…"   7 minutes ago       Exited (137) 2 minutes ago                       nginx
261b98debc89        getumbrel/middleware:v0.1.7          "docker-entrypoint.s…"   7 minutes ago       Exited (137) 2 minutes ago                       middleware
85a68d5945e3        getumbrel/manager:v0.2.7             "docker-entrypoint.s…"   7 minutes ago       Exited (137) 2 minutes ago                       manager
c59d8f050699        getumbrel/frontail:v4.9.1            "/frontail/docker-en…"   12 minutes ago      Exited (137) 2 minutes ago                       frontail
c6c30470f4e1        getumbrel/dashboard:v0.3.13          "docker-entrypoint.s…"   12 minutes ago      Exited (137) 2 minutes ago                       dashboard
a2b6868786a0        getumbrel/tor:v0.4.1.9               "tor"                    12 minutes ago      Exited (137) 2 minutes ago                       tor
umbrel@umbrel:~ $ wget https://raw.githubusercontent.com/getumbrel/umbrel/4603b0357ae5f118847787cd8f4032b99e7947da/scripts/debug  -O ~/umbrel/scripts/debug && chmod +x ~/umbrel/scripts/debug && ~/umbrel/scripts/debug --upload
/home/umbrel/umbrel/scripts/debug: Input/output error
umbrel@umbrel:~ $ ls
umbrel
umbrel@umbrel:~ $ cd umbrel/
umbrel@umbrel:~/umbrel $ ls
ls: reading directory '.': Input/output error
```

hitchcott, 2020-12-22:
Update: 

I'm having success with using the USB2 port instead of the (blue) USB3 port on the Pi. No other changes, and no I/O errors anymore.

jonsyu, 2021-12-16:
Closing due to inactivity, let us know on https://community.getumbrel.com if the issue still exists.

closed 2021-12-16 by jonsyu