getumbrel/umbrel-os #162: [FEATURE REQUEST] Add boot from USB/SSD support

reinto, 2021-01-11:
Please add support for booting from USB. I flashed the 1TB SSD with the Umbrel-OS image but it fails to start properly and I receive the ERR_CONNECTION_REFUSED error in the browser.

On screen, I see the following service failing:
`[FAILED] Failed to starte External Storage Mounter.
See 'systemctl status umbrel-external-storage.service' for details.
[DEPEND] Dependency failed for Umbrel Startup Service.
[DEPEND] Dependency failed for Connection Details.`

At the bottom the cursor is just blinking away at:
`umbrel login:`

AaronDewes, 2021-01-12:
@reinto  You can try if https://github.com/getumbrel/umbrel/suites/1809392088/artifacts/34767027 works.

reinto, 2021-01-12:
Thanks, I will! I'll update here how that turns out.

Can I ask what the main motivation is to use different devices (SD card and SSD disk) as opposed to running all from a single disk? That is what it says on the pull request. 
My motivation not to use an SD card is durability and speed, although perhaps those are not that important if the SD card only holds the OS and, for example, Page files and other high I/O stuff are turned off.

AaronDewes, 2021-01-12:
The OS on the SD allows easily reflashing the OS in case anything goes wrong without loosing all data, because all data except the docker container images are on the SSD.

reinto, 2021-01-12:
I guess you're right! If it indeed saves your data and let's you re-flash the image to SD, that is a major win. Will it save both the synced full node and any open LN channels?

I can test the bits, if that would help. However, I will probably end up running the OS on an SD card for easy recovery now.

AaronDewes, 2021-01-12:
Yes, everything of the data is on the SSD, only the OS is on the SD

closed 2021-01-12 by reinto

dskvr, 2022-11-07:
> The OS on the SD allows easily reflashing the OS in case anything goes wrong without loosing all data, because all data except the docker container images are on the SSD.

IMO this is not a good tradeoff considering exceptionally poor performance reported frequently by community and SD card corruption for so much as a power outage. The SSD can be partitioned, and the Umbrel could live in one of the partitions. 

1. Would operate faster.
2. Would be cheaper to fix.
3. Would not take 1 hour to load the admin interface (yes, official PS and bench-tested USB3 cable)