getumbrel/umbrel-os #165: create entry in /etc/fstab

phenomeus, 2021-01-24:
hi everybody :) 

would it be possible to add the connected hard drive to /etc/fstab automagically by uuid after the first boot? 
I did the mistake and added my second USB thumb into the pi4 and umbrel couldn't mount sda1 to /mnt/data correctly.

after that I tried to add the entry by hand into /etc/fstab but I fckd up and it was only mounted read only… 🤦 

so umbrel called back to write / sync the btc chain to the SD card (!) and I catched that only because I already was syncing 5 years and the process as the complete "setup flow" was restarted and the chain restarted to sync again. 

i fixed it now and the chain is syncing again from 5y upwards. but to keep others from the pain I experienced I would suggest to add the entry into /etc/fstab so every one can be sure that SDA1 is mounted correctly to /mnt/data  

(any tipps to clean up my SD card… ? blocks and chain state are already removed, but other files in ~/umbrel/… ? )

lukechilds, 2021-01-26:
We don't want want Linux to handle mounting the block device automatically, we want to do it manually so we can do our own checks. This is all handled via the `umbrel-external-storage` service.

Currently a limitation of our implementation is that you can't have multiple block devices plugged in at the same time.

Did you manually start Umbrel via SSH? If Umbrel detects two block devices it will refuse to start to prevent writing to the SD card.

phenomeus, 2021-01-26:
yeah I did manually start it, and it apparently wrote to my SD card.