getumbrel/umbrel-os #166: new umbrel-details for 3.5" touch LCD

phenomeus, 2021-01-29:
inspired and coming from raspiblitz, I have a 3.5 touch LCD attached on my pi4 running umbrel :) 

being to stupid to code it myself I can give some modified code snippets. hope someone can adjust it and add it to umbrel for people with this display attached. ["Waveshare 3.5 inch Resistive Touch Screen TFT LCD Display XPT2046 125MHz High-Speed SPI 480x320"](https://www.amazon.de/Waveshare-Resistive-High-Speed-Raspberry-Directly-pluggable/dp/B07L42Z79L/ref=sr_1_1?__mk_de_DE=ÅMÅŽÕÑ&dchild=1&keywords=Waveshare+3.5+inch+Resistive+Touch+Screen+TFT+LCD+Display+XPT2046&qid=1611957312&sr=8-1)

`
##!/bin/bash

# set colors
color_red='\033[0;31m'
color_green='\033[0;32m'
color_amber='\033[0;33m'
color_blue='\033[0;94m'
color_gray='\033[0;37m'

clear
printf "
${color_blue}
${color_blue}
${color_blue}             .,cdkOOOkdc'             ${color_gray}
${color_blue}         .ckXWMMMMWNWMMMMWKx;         ${color_gray}
${color_blue}       lKMMKxc'.      ..,lkNMWk'      ${color_gray}
${color_blue}     cNMWx.                 ;KMM0.    ${color_gray}
${color_blue}    OMMO.                     :NMWc   ${color_gray}
${color_blue}   0MMX:;;,.    .';;;'.   .',,,lWMMc  ${color_gray}
${color_blue}  dMMMMMMMMMWKKWMMMMMMMW0NMMMMMMMMMW, ${color_gray}
${color_blue}  l0k;.   .,lkkl'.   .;okx:.   .'lOO, ${color_gray}
${color_blue}                                      ${color_gray}
${color_blue}             ;;        .;;            ${color_gray}
${color_blue}            xMMx       OMMl           ${color_gray}
${color_blue}            kMMk       KMMl           ${color_gray}
${color_blue}            kMMk       KMMl           ${color_gray}
${color_blue}            xMMk       KMMl           ${color_gray}
${color_blue}            oMMO       KMMl           ${color_gray}
${color_blue}            .XMMk;...:OMMK.           ${color_gray}
${color_blue}              cOWMMMMMNO:             ${color_gray}
${color_blue}                 ,clc,                ${color_gray}
${color_blue}
$lastLine
" \

# EOF
`

louneskmt, 2021-01-29:
Check this: https://github.com/EddWills95/UmbrUI

bart, 2021-08-30:
I updated `umbrel-details` script to display RaspiBlitz like information. See https://github.com/bart/umbrel-status-screen