getumbrel/umbrel-os #169: samsung nvme disks supported

rojomisin, 2021-02-08:
I got a 1tb samsung nvme m.2 ssd and have it in a unitek enclosure, umbrel fails to see it

```
root@umbrel:~# systemctl status umbrel-external-storage.service 
● umbrel-external-storage.service - External Storage Mounter
   Loaded: loaded (/etc/systemd/system/umbrel-external-storage.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-02-08 22:42:50 UTC; 33min ago
  Process: 339 ExecStart=/home/umbrel/umbrel/scripts/umbrel-os/external-storage/mount (code=exited, status=32)
 Main PID: 339 (code=exited, status=32)

Feb 08 22:42:41 umbrel external storage mounter[339]:         102400000, 214990848
Feb 08 22:42:41 umbrel external storage mounter[339]: [65B blob data]
Feb 08 22:42:41 umbrel external storage mounter[339]: [62B blob data]
Feb 08 22:42:45 umbrel external storage mounter[339]: Creating journal (262144 blocks): done
Feb 08 22:42:50 umbrel external storage mounter[339]: [99B blob data]
Feb 08 22:42:50 umbrel external storage mounter[339]: Mounting partition...
Feb 08 22:42:50 umbrel external storage mounter[339]: mount: /mnt/data: special device /dev/sda1 does not exist.
Feb 08 22:42:50 umbrel systemd[1]: umbrel-external-storage.service: Main process exited, code=exited, status=32/n/a
Feb 08 22:42:50 umbrel systemd[1]: umbrel-external-storage.service: Failed with result 'exit-code'.
Feb 08 22:42:50 umbrel systemd[1]: Failed to start External Storage Mounter.
root@umbrel:~# lsblk 
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 29.6G  0 disk 
├─mmcblk0p1 179:1    0  256M  0 part /boot
└─mmcblk0p2 179:2    0 29.4G  0 part /

```

rojomisin, 2021-02-09:
ssd: https://www.bhphotovideo.com/c/product/1398241-REG/samsung_mz_v7e1t0bw_1tb_970_evo_nvme.html/?ap=y&smp=y&msclkid=3c5e1a9a65f91b2fa9857abbb25fa70f

enclosure: https://www.unitek-products.com/collections/hdd-ssd-enclosure/products/usb3-1-gen2-type-c-to-m-2-ssd-nvme-enclosure-black

i ran `apt install nvme-cli` hoping it would install a driver that could see the drive, but no luck so far.

benpbolton, 2021-02-12:
I can't directly test without the hardware, but is it possible the enclosure isn't getting enough power via the pi? 

Can you test with a powered usb hub in the middle?

jonsyu, 2021-12-16:
Closing due to inactivity, let us know on https://community.getumbrel.com if the issue still exists.

closed 2021-12-16 by jonsyu