getumbrel/umbrel-os #177: Required dependencies not installed

rstmsn, 2021-02-16:
Having trouble compiling from source. I'm using a Raspberry PI to compile and build.

`sudo ./build.sh 
Required dependencies not installed

This can be resolved on Debian/Raspbian systems by installing:
docker`

When I 'apt install docker', you can see that it is arleady installed:

`sudo apt install docker
Reading package lists... Done
Building dependency tree       
Reading state information... Done
docker is already the newest version (1.5-2).`

Is this a versioning issue, perhaps? 
Any help appreciated.

rstmsn, 2021-02-16:
I reinstalled docker using the official install script:
`curl -sSL https://get.docker.com | sh`
as opposed to apt get.
I also added my user to the docker user group:
`sudo usermod -aG docker pi`

and the build is now proceeding!

closed 2021-02-16 by rstmsn

AaronDewes, 2021-02-17:
> docker is already the newest version (1.5-2).

The text is just incorrrect, the package "docker" isn't actually docker, docker is in docker.io, I'll fix that