getumbrel/umbrel-os #182: bech32 addresses?

zatJUM, 2021-02-23:
Great work!
I'm trying to find the balance of a bc1 address, and I came to the point they're not supported yet on umbrel.
Is there is reason why "addresstype=bech32" is not set in bitcoin.conf?

downet, 2021-02-23:
When I choose Bitcoin and deposit, I see bc1 address..

zatJUM, 2021-02-23:
Yes, so do I :thinking: 
So it must be a mempool config problem?

downet, 2021-02-23:
But which problem do you have?

zatJUM, 2021-02-23:
I cannot see a bc1 address in the mempool app:
"Error: Connection to server lost, please retry"
But if I enter a txid, server answers.

downet, 2021-02-23:
For me it´s working well.

[image]

Can you try to run via ssh following command: `docker logs --tail=200 -f mempool_api_1` and try to search bc1 address, if you will see some error :)

zatJUM, 2021-02-23:
I see nothing strange

downet, 2021-02-23:
Means you do not see any log or any error?

zatJUM, 2021-02-23:
Many things like these:```Feb 23 20:45:15 [80] DEBUG: Updating mempool
Feb 23 20:45:15 [80] DEBUG: Fetched transaction 1 / 12
Feb 23 20:45:15 [80] DEBUG: Fetched transaction 2 / 12
Feb 23 20:45:15 [80] DEBUG: Fetched transaction 3 / 12
Feb 23 20:45:15 [80] DEBUG: Fetched transaction 4 / 12
Feb 23 20:45:15 [80] DEBUG: Fetched transaction 5 / 12
Feb 23 20:45:15 [80] DEBUG: Fetched transaction 6 / 12
Feb 23 20:45:15 [80] DEBUG: Fetched transaction 7 / 12
Feb 23 20:45:15 [80] DEBUG: Fetched transaction 8 / 12
Feb 23 20:45:15 [80] DEBUG: Fetched transaction 9 / 12
Feb 23 20:45:15 [80] DEBUG: Fetched transaction 10 / 12
Feb 23 20:45:15 [80] DEBUG: Fetched transaction 11 / 12
Feb 23 20:45:15 [80] DEBUG: Fetched transaction 12 / 12
Feb 23 20:45:16 [80] DEBUG: New mempool size: 18137 Change: 12
Feb 23 20:45:16 [80] DEBUG: Mempool updated in 0.917 seconds
Feb 23 20:45:18 [80] DEBUG: Updating mempool
Feb 23 20:45:18 [80] DEBUG: Fetched transaction 1 / 11
Feb 23 20:45:18 [80] DEBUG: Fetched transaction 2 / 11```

downet, 2021-02-23:
Did you tried to reinstall it? That is last thing which is in my mind :)

zatJUM, 2021-02-24:
Yes! In fact I had already removed and reinstalled, but this time, it worked :-)
Thanks for your help

closed 2021-02-24 by zatJUM