getumbrel/umbrel-os #190: Error loading shared library libsqlite3.so.0: No such file or directory (needed by /usr/local/bin/bitcoind)

x775, 2021-02-28:
Upon losing a carrier and immediately before acquiring a new carrier, the following errors occur.

```
umbrel dhcpcd[700]: veth32bd5b5: removing interface
umbrel dhcpcd[700]: veth3be4691: carrier lost
umbrel bitcoin[729]: Error loading shared library libsqlite3.so.0: No such file or directory (needed by /usr/local/bin/bitcoind)
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_column_text: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_close: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_column_int: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_reset: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_clear_bindings: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_open_v2: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_column_bytes: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_shutdown: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_exec: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_config: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_backup_step: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_db_readonly: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_backup_init: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_finalize: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_get_autocommit: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_step: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_backup_finish: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_errstr: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_errmsg: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_column_blob: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_initialize: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_libversion: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_prepare_v2: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_bind_blob: symbol not found
umbrel bitcoin[729]: Error relocating /usr/local/bin/bitcoind: sqlite3_open: symbol not found
umbrel dhcpcd[700]: veth3be4691: carrier acquired
```

I suspect this is due to libsqlite3 i386 missing, and can be fixed by adding 

`sudo apt install libsqlite3-0:i386`

or merely `libsqlite3-0:i386` to the appropriate list of packages. 

I am not sure which one that is, hence an issues instead of a PR.

If is already being installed, there are issues and a `sudo apt install --reinstall libsqlite3-0:i386` should do the trick.

AaronDewes, 2021-02-28:
> i386 

That would be the version for Intel 32 bit computers, and not for arm64 RPis.
Also, this needs to be present inside one container, not on the host. This bug is already reported, we'll provide a fix soon.

x775, 2021-02-28:
> 
> 
> > i386
> 
> That would be the version for Intel 32 bit computers, and not for arm64 RPis.
> Also, this needs to be present inside one container, not on the host. This bug is already reported, we'll provide a fix soon.

Of course, my bad.

Sounds good with the fix.

closed 2021-02-28 by x775