getumbrel/umbrel-os #191: Fix PREFIX value for mempool app API

zatJUM, 2021-03-01:
I'm trying to use the API of my running node, but I think there's a problem with the configuration file of the mempool app:
(the address is not mine)
```bash
curl -s http://umbrel.local:3006/api/address/bc1qh9c7xfz0eu3kvqtz6lgm0v0nqvcg5nkrwn3k6m/txs/mempool
<!DOCTYPE html>
<html lang="en">
<head>
<meta charset="utf-8">
<title>Error</title>
</head>
<body>
<pre>Cannot GET /api/v1/address/bc1qh9c7xfz0eu3kvqtz6lgm0v0nqvcg5nkrwn3k6m/txs/mempool</pre>
</body>
</html>
```
But syntax is ok and works fine with mempool.space site:
```bash
curl -s https://mempool.space/api/address/bc1qh9c7xfz0eu3kvqtz6lgm0v0nqvcg5nkrwn3k6m/txs/mempool
[]
```

Have you noticed the `v1` added in the url on my node?

zatJUM, 2021-07-16:
Update:
It seems related to the requested endpoint, as this command works fine:
```curl -s http://umbrel.local:3006/api/address/bc1qh9c7xfz0eu3kvqtz6lgm0v0nqvcg5nkrwn3k6m/txs```

jonsyu, 2021-12-16:
Closing due to inactivity, let us know on https://community.getumbrel.com if the API is still giving you trouble.

closed 2021-12-16 by jonsyu