getumbrel/umbrel-os #195: unable to setup umbrel because of CORS issue

lvnilesh, 2021-03-03:
My router dynamically assigns DNS names to machines so this new raspi4 got assigned a DNS name something like raspi4.workplace.domain.com

Now I can not set up umbrel because of a CORS issue.

https://github.com/getumbrel/umbrel-os/issues/63

epson121, 2021-04-17:
I managed to get over the issue by updating `/etc/hosts` file, and adding this:

`192.168.1.10 umbrel.local`

change the IP to the one used by your umbrel. This allowed me to go over the CORS issue

raymondclowe, 2021-07-22:
I had the same problem running it on WSL2 on Win10.

Editing the hosts file alone did not fix it, probably because I was using port 12345 instead of 80.

But also add to the scripts/start file and then it was ok.  As follows:

`DEVICE_HOSTS="http://${DEVICE_IP},http://${DEVICE_HOSTNAME}.local,https://${DEVICE_HOSTNAME}.local,http://${DEVICE_HOSTNAME},https://${DEVICE_HOSTNAME},http://umbrel.local:12345"`

Then when I accessed it from `http://umbrel.local:12345` it worked.

lvnilesh, 2021-07-25:
Thanks. 

I edited `./scripts/start` to include `raspi4.workplace.domain.com` and now it is working.

closed 2021-07-25 by lvnilesh