getumbrel/umbrel-os #196: Fresh umbrel set up. Dead in 40 hours.

lvnilesh, 2021-03-04:
How do I identify the root cause of this death?

```
umbrel@umbrel:~ $ docker ps -a
CONTAINER ID   IMAGE                                COMMAND                  CREATED        STATUS                     PORTS                                                                                           NAMES
6cecaa7dafc6   getumbrel/neutrino-switcher:v1.2.0   "/usr/local/bin/swit…"   45 hours ago   Exited (255) 5 hours ago                                                                                                   neutrino-switcher
e3477f8b11e0   getumbrel/middleware:v0.1.9          "docker-entrypoint.s…"   45 hours ago   Exited (255) 5 hours ago   3006/tcp                                                                                        middleware
8a1138bc6a47   getumbrel/electrs:v0.8.6             "electrs"                45 hours ago   Exited (255) 5 hours ago   4224/tcp, 0.0.0.0:50001->50001/tcp                                                              electrs
40aa67c969b5   lncm/bitcoind:v0.21.0                "bitcoind -zmqpubraw…"   45 hours ago   Exited (255) 5 hours ago   8080/tcp, 8332/tcp, 18332-18333/tcp, 18443-18444/tcp, 28332-28333/tcp, 0.0.0.0:8333->8333/tcp   bitcoin
ee5ca327c3d7   lncm/lnd:v0.12.1                     "/usr/local/bin/lnd"     45 hours ago   Exited (255) 5 hours ago   0.0.0.0:8080->8080/tcp, 0.0.0.0:9735->9735/tcp, 0.0.0.0:10009->10009/tcp, 9911/tcp              lnd
3c0e078160b9   nginx:1.17.8                         "nginx -g 'daemon of…"   45 hours ago   Exited (255) 5 hours ago   0.0.0.0:80->80/tcp                                                                              nginx
a7ae60392fcb   getumbrel/manager:v0.2.10            "docker-entrypoint.s…"   45 hours ago   Exited (255) 5 hours ago   3006/tcp                                                                                        manager
8f7239148d12   getumbrel/dashboard:v0.3.17          "docker-entrypoint.s…"   45 hours ago   Up 5 hours                 3004/tcp                                                                                        dashboard
00f61f8ba4a1   lncm/tor:0.4.4.7                     "tor"                    45 hours ago   Exited (255) 5 hours ago   9051/tcp, 127.0.0.1:9050->9050/tcp, 29050-29051/tcp                                             tor
04de1807ac14   getumbrel/frontail:v4.9.1            "/frontail/docker-en…"   45 hours ago   Up 5 hours                 9001/tcp                                                                                        frontail
umbrel@umbrel:~ $
```

lvnilesh, 2021-03-05:
This is the second time I notice umbrel died after about 40 hours of initial setup.

AaronDewes, 2021-03-05:
Can you share the links of `~/umbrel/scripts/debug --upload`?

lukechilds, 2021-03-05:
It's most likely shutdown in an attempt to prevent filesystem corruption after noticing instability.

Are you using the official Raspberry Pi power supply?

Nathanw1014, 2021-03-06:
Could you please define 'died'?
Is it just not accessible on the IP/umbrel.local,
does it respond locally when you plug in a keyboard/screen ect?

I may be having similar issues since the latest update, just want to check if its the same.

lvnilesh, 2021-03-07:
@lukechilds Thanks for the pointer. I was using Apple official MacBook Pro USB-C Charger. I will purchase official power supply now.

lvnilesh, 2021-03-07:
> Could you please define 'died'?
> Is it just not accessible on the IP/umbrel.local,
> does it respond locally when you plug in a keyboard/screen ect?
> 
> I may be having similar issues since the latest update, just want to check if its the same.

I can ssh. 
All `docker processes` dead as shown in my original post above.
I can cd into `/mnt/data`
`ls` in there does not produce meaningful results.

closed 2021-03-07 by lvnilesh