getumbrel/umbrel-os #199: Can't access Umbrel.local

ChrisWeisler, 2021-03-14:
I can no longer access umbrel.local/dashboard but AngryScanner is still able to see my Umbrel's IP address.  I've tried using SSH and was able to log in via the mac os terminal. However, the debug scripts I found (from dec 2020) and tried, no longer work.

Over the last two weeks, the umbrel.local dashboard would start losing connection, but I found that a simple unplug and reset of my Raspi would always get me back to a working login.  unfortunately, that no longer works.

I have limited programming abilities, but any help would be greatly appreciated.
Cheers.

AaronDewes, 2021-03-14:
Can you contact me on Telegram? t.me/AaronDewes. Chatting there makes debugging easier.

AaronDewes, 2021-03-14:
Also, please only report bugs in Umbrel on GitHub, and ask us on Telegram for anything else. Also, running the debug script with wget ... breaks it (And also makes it impossible to use it in the future without a few manual fixes). Please keep in mind that Umbrel is still in beta, and things keep changing and evolving, so older guides might not work today. info.umbrel.tech always has working guides for the latest version, but older comments might not work on the latest Umbrel.

ChrisWeisler, 2021-03-15:
Got it sorted by reflashing. Thanks for help Aaron.

closed 2021-03-15 by ChrisWeisler