getumbrel/umbrel-os #200: It cannot be started after a power failure

RomanD9, 2021-03-15:
Today there have been power outages. Unable to start Umbrel. I haven't finished verifying the blockchain.

kieselbert, 2021-03-21:
similar issue here, after a power outage (accidentally unplugged) is the node faulty. It is not possible to login, it stucks in verifying blockchain.

ThirdArrow, 2021-05-01:
Similar here as well: was still in the process of validating, lost power, now unable to start Umbrel.
Umbrel.local not showing up on IP scanner, RasPi showing intermittent green lights with solid red. SSD lights up briefly but after a minute or two goes dark.

Going to try and flash the OS onto a new SD and see if that does the trick but it will be a few days until I can order one.

HenryWait, 2021-06-11:
I have had no success powering on my Umbrel after shutting it down and discovered it will only boot with a display connected. 
This is an issue with the Raspi 4 when running headless, the earlier Raspi models would fallback to composite output when no display is connected and boot fine, the pi 4 however fails to boot if no HDMI mode is specified in /boot/config.txt and no display is connected. 

The state of my /boot/config.txt when the issue occured:
umbrel@umbrel:~ $ cat /boot/config.txt | grep hdmi
#hdmi_safe=1
# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1
#hdmi_group=1
#hdmi_mode=1
#hdmi_drive=2
#config_hdmi_boost=4

The fix is adding a HDMI mode to /boot/config.txt 
# uncomment to force a specific HDMI mode (this will force 1080p)
hdmi_group=2
hdmi_mode=82

Uncommenting #hdmi_force_hotplug=1  instead should also allow it to boot headless.

jonsyu, 2021-12-16:
Closing due to inactivity, let us know on https://community.getumbrel.com if the issue still exists.

closed 2021-12-16 by jonsyu