getumbrel/umbrel-os #203: moving/migrating wallets + payment channels from one umbrel to another

jlatone, 2021-03-25:
hi, i want to move the bitcoin + lightning wallets and lightning payment channels from one umbrel to another. 

how do i do it? ideally, just copy/move a few files, perhaps issue a `lndcli` command? :)

here's more detail:

i have ~100k sats in the umbrel bitcoin wallet 

[image]

and ~100k sats in the umbrel lightning wallet 

[image]

if it helps, here are the umbrel files with "wallet" in the name:

```
-rw------- 1 ec2-user ec2-user 1445888 Mar 25 04:52 ./bitcoin/wallet.dat
-rw------- 1 ec2-user ec2-user       0 Mar 24 21:11 ./bitcoin/.walletlock
-rw------- 1 ec2-user ec2-user 2097152 Mar 25 16:10 ./lnd/data/chain/bitcoin/mainnet/wallet.db
-rw-r--r-- 1 ec2-user ec2-user     114 Dec  8 21:08 ./lnd/data/chain/bitcoin/mainnet/walletkit.macaroon

```

and, i want to move the following lightning payment channel (which happens to be to getumbrel), 
[image]

i don't want to copy the entire umbrel because I don't want to copy the bitcoin blockchain since it's already sync'd on the destination umbrel (and it's too costly to move).

thanks for any help!

--joe

downet, 2021-03-25:
Hi,

My guess would be:
1) on new machine, use same seed as on previous - that would solve Bitcoin address
2) stop new and old umbrel and copy whole /mnt/data/umbrel/lnd/ folder from old to new and start new.. NEVER EVER START THE OLD ONE!
3) Profit :)

Jaxo, 2021-03-25:
> 1. on new machine, use same seed as on previous - that would solve Bitcoin address
> 3. Profit :)

Thanks! To help ensure I profit, instead of loss, how do I do #1? I.e., where do I enter a different seed or enter recovery mode, since the destination umbrel is already running?

downet, 2021-03-25:
If second umbrel (destination one) is running, you can reset it via https://info.umbrel.tech/troubleshooting.html#reset-your-umbrel-user-data-if-you-lost-your-password

Jaxo, 2021-03-26:
> https://info.umbrel.tech/troubleshooting.html#reset-your-umbrel-user-data-if-you-lost-your-password

It all worked! 
Thanks!
🥇

jonsyu, 2021-12-16:
Closing due to inactivity

closed 2021-12-16 by jonsyu