getumbrel/umbrel-os #204: User already exists message when recovering wallet

Kryptophreak, 2021-03-30:
When recovering from seed, if user goes past the step after entering password and verifying it, recover wont work.
Even though there is a back button it does not help.
Once you go past and a seed is generated, if you go back, it is like the system is trying to generate a new user with your seed you want to recover with.
The only way I have found to go back is to re flash the SD card, reformat the SSD and start all over.
If you want to recover form seed, when you get to the start page, hit RECOVER, enter the correct seed for the wallet to recover and it works fine.
If you go past this step, you have to start all over.
I can imagine  anyone trying to recover a wallet with significant funds on it will have rather high anxiety levels if they encounter the above conditions.

tychsen, 2021-05-05:
bump for this one!

Synesso, 2021-07-06:
There is no "recover" button visible until after the password has been entered and confirmed.

Bucherbp, 2021-08-01:
Hey @Kryptophreak , I just so happen to be one of those high anxiety users. While the funds I had on the node weren’t significant, it does have me feeling like I can’t actually trust that I could recover an umbrel node if I did depend on it for more of my bitcoin usage.

Anyway, I am having trouble understanding the alternative procedure here that would make it work. From the umbrel start page, the first thing it does is ask you for a name, then a password. I don’t think I see an option to recover until after the name and password are chosen (though I am away from the setup right now, and can’t verify).

To be clear, what you’re saying is to choose RECOVER right from the first start page, before entering a name and password?

Synesso, 2021-08-01:
@Bucherbp I was in the same boat as you. Strangely, the work-around was to create the exact same password before recovering. If you create with a different password you need to wipe the drive and start again.

Bucherbp, 2021-08-01:
Hmm, I am relatively sure I am using the same password I had been using the node before recovering. At the very least, I am using the same password I was using when the node went down. I don’t remember what password I used when I originally created the node, it may or may not have been the same.

I am running into the same problem, over and over, even when creating a fresh seed for testing. I spin up the umbrel, and create a new user and seed. Flash and format. Immediately use that new seed I just created, and I keep getting the “user already exists” error. I’ll pay more attention to the password when trying this again.

Even if that is the case, this seems like a very insecure way to have your node work. If the derivation of your seed, and therefore the recovery, depend on the password chosen when you first start, then Umbrel should not allow you to change your password, ever. A user would assume this was safe, forget the original password, and never be able to recover their funds.

Synesso, 2021-08-01:
100% agree. This is a significant bug I'd argue.

Bucherbp, 2021-08-02:
Re-reading @Kryptophreak ’s post, I think i now understand what he was saying. He is saying that if you go past the page where you can choose recover, initiating a new seed creation, then go back, a recovery can’t be completed.

Mine is a bit different, as I am not doing that. But, I still can’t recover any seed.

Bucherbp, 2021-08-02:
Hey @Kryptophreak  and @Synesso , it turns out that my problem was precisely as @Kryptophreak describes. I did have to hit the back button at some stage in my procedure. This is because my computer keep trying to suggest a password, which I would then delete, and enter my own. But this would ultimately cause an issue, so I would have to go back to the first password entry screen, type it again, then confirm.

I switched browsers so that this didn’t happen anymore, no back button, no issue restoring the seed. Turns out that back button is the key issue.

jonsyu, 2021-12-16:
Closing due to inactivity, let us know on https://community.getumbrel.com if the issue still exists.

closed 2021-12-16 by jonsyu