getumbrel/umbrel-os #212: The ssh password is reset to default after upgrading from 0.3.8 to 0.3.9

lestephane, 2021-04-24:
I used to be able to SSH into the box using the same password as the web dashboard.
I needed to SSH into the box after flashing 0.3.9 when I realized that 0.3.9 did not resolve [this other issue](https://github.com/getumbrel/umbrel/issues/613).

But I cannot login over SSH using the web dashboard password anymore. I had to use the default password. 

How does that make any sense? There is no note about this in (non-existing) release notes.

AaronDewes, 2021-04-24:
We did not reset the password, also release notes exist here on GitHub. A reflash however resets the password until you login into the dashboard, this has nothing to do with the update.

lestephane, 2021-04-24:
I can't login to the dashboard because of issue #613. 
So this behaviour still makes no sense to me.

AaronDewes, 2021-04-24:
It is **impossible** to do this securely, because then we would have to store your password in plain text on the SSD, which is very insecure, and it would also make SSH access impossible if you lost your password, with this, you can access it by reflashing.

lestephane, 2021-04-24:
I can't login to the dashboard because of issue #613.
WHAT DO I DO?

AaronDewes, 2021-04-24:
> WHAT DO I DO?

STOP SCREAMING!

But you should join us on Telegram to get help.

lestephane, 2021-04-24:
Not moving the discussion to Telegram, because any tribal knowledge uncovered there will scroll out of sight. And chats are not encrypted, and there is too much noise, with people having to get banned constantly for peddlng shitcoin scams.

I would go to Telegram, then you'd help me, and then anyone with the same problem in one week or one month has to ask again, and uncover the tribal knowledge again.

This issue should at the very least explain what the user is to do when :

1) login to the web interface is not possible
2) reflashing means the default password is now in use

Am I supposed to
1) login using the default password
2) change the SSH password to the same as the dashboard 

Would that be sufficient to reestablish the same configuration that existed when I had 0.3.8? 
Or are there other aspects to consider?

This is what I meant with WHAT DO I DO? I was simulating the non tech user who's going to come to you with zero clue. 

I'm a tech user so I'll figure it out, but please please please, see this question from the angle of one of the thousands of non tech users that will use umbrel and encounter the same problem.

AaronDewes, 2021-04-24:
The first step is fixing the web interface, then everything else is done automatically. Please run `sudo ~/umbrel/scripts/debug --upload` and send me the link it generates.

AaronDewes, 2021-04-24:
Please do a manual update to 0.3.9 as described on info.umbrel.tech.

AaronDewes, 2021-04-24:
There's a bug I just found in 0.3.9 which means a manual update is required, but we're working on it.

lestephane, 2021-04-24:
For the #613 issue, I posted the debug upload to #613, where it belongs.

This issue is about whether, after reflashing (and therefore the password being reset), changing the SSH password from default to <the same as Web dashboard> *manually* is the same as logging in successfully to the dashboard.

I will check the info.umbrel.tech page you're referring to (https://info.umbrel.tech/troubleshooting.html#manually-updating-umbrel) and another one Luke pointed me to (https://github.com/getumbrel/umbrel/blob/master/SECURITY.md) to try to figure that out on my own.

lestephane, 2021-04-24:
> There's a bug I just found in 0.3.9 which means a manual update is required, but we're working on it.

If you have an issue number, I'm interested.

AaronDewes, 2021-04-24:
> It's not a GitHub issue yet.

AaronDewes, 2021-04-24:
> This issue is about whether, after reflashing (and therefore the password being reset), changing the SSH password from default to the same as Web dashboard manually is the same as logging in successfully to the dashboard.

Yes, it is

lukechilds, 2021-04-24:
>This is what I meant with WHAT DO I DO? I was simulating the non tech user who's going to come to you with zero clue.

I understand your frustration @lestephane.

Just to clarify how this works, it's only very brief. Once you re-flash the SD card, the password is reset because it can't possibly know your password. When you next login to the dashboard, which would likely be very soon after re-flashing, if there is a successful login, your system/SSH password will be updated to use that password. So Umbrel is (almost) instantly updated to use your supplied password again after a re-flash, most users would never experience the password reset.

I understand you can't complete the login process due to the LND loading screen, however the latest release has resolved that issue. The login process is no longer dependent on LND being available, so there should not be a scenario where a user re-flashes but is unable to login.

lukechilds, 2021-04-24:
>I realized that 0.3.9 did not resolve this other issue.

Are you saying you're on 0.3.9 but you're still seeing the "Loading LND..." text? That shouldn't be the case, that could is completely removed. Can you try to refresh the browser?

lukechilds, 2021-04-24:
>There is no note about this in (non-existing) release notes.

For release notes check the Umbrel releases: https://github.com/getumbrel/umbrel/releases/tag/v0.3.9

We have a strict policy of one change = one PR = one merged commit. So git log is our release notes. You can see a descriptive overview of the changes between each release by comparing the two tags, and follow the PR links for more detailed information on each change: https://github.com/getumbrel/umbrel/compare/v0.3.8...v0.3.9

AaronDewes, 2021-04-24:
> Are you saying you're on 0.3.9 but you're still seeing the "Loading LND..." text? That shouldn't be the case, that could is completely removed. Can you try to refresh the browser?

This is the bug I told you about a few minutes ago, where a reflash doesn't update.

lukechilds, 2021-04-24:
Ah, of course!

Ok, looking into that on priority now.

lukechilds, 2021-04-24:
@lestephane this should work, are you able to try re-flashing your SD card with this build: https://github.com/getumbrel/umbrel-os/releases/tag/v0.3.9-patch-2

lestephane, 2021-04-24:
> Are you saying you're on 0.3.9 but you're still seeing the "Loading LND..." text?

I do, at the **Safest** security setting. **Safer** security setting goes through.

The error at the safest security setting is below (HAR is attached)

Which brings up an interesting point: If I've doxed my onion service, how do I get Umbrel to generate a new onion service name? Is that something that is currently possible? Manual steps also OK.

[trace.har.zip](https://github.com/getumbrel/umbrel-os/files/6370696/trace.har.zip)

```
TypeError: t.style is undefined
    mi http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    yi http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    M http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    f http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    g http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    f http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    g http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    f http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    g http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    f http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    Aa http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    _update http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    r http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    get http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    nr http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    xn http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    $mount http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    init http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    p http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    f http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    g http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    f http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    g http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    f http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    g http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    f http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    Aa http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    _update http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    r http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    get http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    nr http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    xn http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    $mount http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    init http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    n http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    p http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    f http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    g http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    f http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    Aa http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    _update http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    r http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    get http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    nr http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    xn http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
[... cut ...]
    get http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    run http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    Kn http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    pe http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
    ue http://p73owb3jupqrv5seewqnuyzmzvwxigewwvfwm37xrfqo2dm7aps5hnyd.onion/js/chunk-vendors.9caeae12.js:63
chunk-vendors.9caeae12.js:63:12041
```

lestephane, 2021-04-24:
> @lestephane this should work, are you able to try re-flashing your SD card with this build: https://github.com/getumbrel/umbrel-os/releases/tag/v0.3.9-patch-2

It is probably a good idea for me to have 2 SD cards and flash the next version on the SD card I'm not using, so that I can, if push comes to shove, go back to the old. It seems kind of obvious in retrospect. But it did not occur to me to buy a couple. I only have one. 

In any case, I will re-flash using my one card and report.

lestephane, 2021-04-25:
I reflashed, and it works at *safer *security settings, still unable to get
in at *safest* setting.
That's despite granting the onion service temp permissions in NoScript.

[image: image.png]

lukechilds, 2021-04-25:
>Which brings up an interesting point: If I've doxed my onion service, how do I get Umbrel to generate a new onion service name? Is that something that is currently possible? Manual steps also OK.

```
cd ~/umbrel && rm -rf tor/data/web && docker-compose restart tor && sleep 1 && sudo reboot
```

>I reflashed, and it works at *safer *security settings, still unable to get in at *safest* setting.

Glad it's working for you now! There's really not much reason to use the strict settings in Tor Browser when accessing Umbrel. That setting is to preserve your privacy when accessing untrusted 3rd party websites by limiting browser features. It's an increase in privacy at the cost if usability. Umbrel is your own trusted server, not a third party, so you don't really need privacy from yourself. It's assumed that JavaScript and all browser APIs are fully operational in the Umbrel UI. Tor is providing secure remote access in this use case, not anonymity.

lestephane, 2021-04-25:
> This should do the trick:

Thanks!

lestephane, 2021-04-28:
The recommended steps for people experiencing the problem on 0.3.9 is to flash 0.3.9-patch2 as Luke describes. closing.

closed 2021-04-28 by lestephane