getumbrel/umbrel-os #215: First clean install, but it's not running

maravac, 2021-04-26:
Hi, please, would you try to help me fix my problem or atleast find out what could be wrong wint my stuff? I just tried to run my first umbrel node on Raspberry Pi 4B (8GB) + 1TB M.2 NVMe SSD in  casing connected to the USB3 of my RPi. After I flashed the SD card, inserted into the RPi, I am not able to connect to it via web browser (either via local IP or umbrel.local). But it seems to be running somehow as it responds to ping and I am also able to connect via SSH. Could it be a problem with the M.2 NVMe SSD drive? It looks like, but this is a brand new M.2 NVMe SSD purchased few days ago and it works when I connect it to my Windows PC - I already  tried to reformat it again, but no change. I am using this M.2 SSD enclosure, by my opinion it should be OK (that secon option - NVMe 10Gbps): 
[]https://www.aliexpress.com/item/4001297316228.html?spm=a2g0s.9042311.0.0.28a94c4dDJZecV

Here is my debug log: []https://umbrel-paste.vercel.app/4b4aa9d6-3b30-436f-952e-58a3c1cdfcc3

I will be very thankful for any kind of help, even just tip, what could be wrong.

AaronDewes, 2021-04-26:
This is very likely some kind of hardware issue.

maravac, 2021-04-27:
Do you think it could be the SSD enclosure not compatible? As far as I tried to investigate, this one should be OK.
Also, may it be the M.2 NVMe SSD itself? I have used the ADATA XPG SX8200 Pro. Can this one be the problem? Physically, it works correctly when used with Windows 10 PC. ANd atleast the raspberry could see it somehow as it was formatted as ext4 and naded as "umbrel" (I checked it when I disconnected it from RPi and connected back to my Win 10 PC).
Will you tell, how can I test morw, what exactly causes my problem? Some SSH commands which can give me some guide to find out more, please?
Also, should there be made some additional steps like ass some configurations, eeprom etc. to let my NVMe drive to boot with RPi?
Thank you.

AaronDewes, 2021-04-27:
It's either the SSD or the enclosure. Umbrel has recommended hardware that works, I don't know about other hardware. It's the power consumption, but I don't know if the SSD itself or the enclosure are the root cause.

maravac, 2021-04-27:
OK, where can I find the list of recommended hardware for Umbrel?

lukechilds, 2021-04-28:
You can find a list of recommended hardware that we've tested no our website: https://getumbrel.com/#start

maravac, 2021-04-29:
Hi, I purchased another M.2 NVMe SSD (Samsung 970 EVO) and now it looks a bit different. Now it seems that my RPi properly can see this SSD drive, but there still are some errors. Maybe something with mounting the device or "can't read superblock on /dev/sda1" ...
So again, I can connect via SSH, but I cannot open the umbrel welcome screen in my web browser.
Just to mention, before I made this another try, I completely reflashed the SD card with the latest umbrel imb and started from the scratch.

(Also, yesterday I also tried another SSD external drive, which worked correctly and I even could open the umbrel welcome page in my browser, but this one was SATA SSD in externbal casing for SATA SSD and it was only 250 GB capacity, so I cannot use it because of this low capacity and also I would like to use M.2 NVMe in external enclosure, as I try now...  because it should provide beter speed)

Please can you check it and give me some help?
[](https://umbrel-paste.vercel.app/7ecbeacb-6bc6-414d-b415-3881f1428f52)

Because paste URL for some reason does not contain everything what I see in SSH output, here it is as I copied it from terminal window:

=====================
= Umbrel debug info =
=====================

Umbrel version
--------------
0.3.9

Umbrel OS version
-----------------
v0.3.9-patch-2

Raspberry Pi Model
------------------
Revision        : d03114
Serial          : 10000000c4bb406a
Model           : Raspberry Pi 4 Model B Rev 1.4

Firmware
--------
Feb 25 2021 12:10:40
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start)

Temperature
-----------
temp=41.8'C

Throttling
----------
throttled=0x0

Startup service logs
--------------------
-- Logs begin at Sat 2021-04-24 18:15:58 UTC, end at Thu 2021-04-29 16:10:10 UTC. --
Apr 29 15:45:15 umbrel systemd[1]: Dependency failed for Umbrel Startup Service.
Apr 29 15:45:15 umbrel systemd[1]: umbrel-startup.service: Job umbrel-startup.service/start failed with result 'dependency'.

External storage service logs
-----------------------------
Apr 24 18:16:03 umbrel external storage mounter[365]: Checking USB devices are back...
Apr 24 18:16:03 umbrel external storage mounter[365]: Waiting for USB devices...
Apr 24 18:16:04 umbrel external storage mounter[365]: Waiting for USB devices...
Apr 24 18:16:05 umbrel external storage mounter[365]: Waiting for USB devices...
Apr 24 18:16:06 umbrel external storage mounter[365]: Checking if the device is ext4...
Apr 24 18:16:07 umbrel external storage mounter[365]: Yes, it is ext4
Apr 24 18:16:07 umbrel external storage mounter[365]: Checking if device contains an Umbrel install...
Apr 24 18:16:07 umbrel external storage mounter[365]: No, it doesn't contain an Umbrel install
Apr 24 18:16:07 umbrel external storage mounter[365]: Unmounting partition...
Apr 24 18:16:08 umbrel external storage mounter[365]: Formatting device...
Apr 24 18:16:08 umbrel external storage mounter[365]: /dev/sda: 8 bytes were erased at offset 0x00000200 (gpt): 45 46 49 20 50 41 52 54
Apr 24 18:16:08 umbrel external storage mounter[365]: /dev/sda: 8 bytes were erased at offset 0xe8e0db5e00 (gpt): 45 46 49 20 50 41 52 54
Apr 24 18:16:08 umbrel external storage mounter[365]: /dev/sda: 2 bytes were erased at offset 0x000001fe (PMBR): 55 aa
Apr 24 18:16:08 umbrel external storage mounter[365]: /dev/sda: calling ioctl to re-read partition table: Success
Apr 24 18:16:12 umbrel external storage mounter[365]: mke2fs 1.44.5 (15-Dec-2018)
Apr 24 18:16:12 umbrel external storage mounter[365]: Creating filesystem with 244190208 4k blocks and 61054976 inodes
Apr 24 18:16:12 umbrel external storage mounter[365]: Filesystem UUID: 481e574f-a37e-4435-88de-e3b465ebe2a8
Apr 24 18:16:12 umbrel external storage mounter[365]: Superblock backups stored on blocks:
Apr 24 18:16:12 umbrel external storage mounter[365]:         32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208,
[... 82 lines cut ...]
services.bitcoin.ports contains an invalid type, it should be a number, or an object
services.electrs.ports contains an invalid type, it should be a number, or an object
services.lnd.ports contains an invalid type, it should be a number, or an object
services.lnd.ports contains an invalid type, it should be a number, or an object
services.nginx.ports contains an invalid type, it should be a number, or an object
services.lnd.ports value ['9735:9735', ':', ':'] has non-unique elements
ERROR: The Compose file './docker-compose.yml' is invalid because:
networks.default.ipam.config.subnet is invalid: should use the CIDR format
services.tor.ports contains an invalid type, it should be a number, or an object
services.bitcoin.ports contains an invalid type, it should be a number, or an object
services.electrs.ports contains an invalid type, it should be a number, or an object
services.lnd.ports contains an invalid type, it should be a number, or an object
services.lnd.ports contains an invalid type, it should be a number, or an object
services.nginx.ports contains an invalid type, it should be a number, or an object
services.lnd.ports value ['9735:9735', ':', ':'] has non-unique elements
ERROR: The Compose file './docker-compose.yml' is invalid because:
networks.default.ipam.config.subnet is invalid: should use the CIDR format
services.tor.ports contains an invalid type, it should be a number, or an object
services.bitcoin.ports contains an invalid type, it should be a number, or an object
services.electrs.ports contains an invalid type, it should be a number, or an object
services.lnd.ports contains an invalid type, it should be a number, or an object
services.lnd.ports contains an invalid type, it should be a number, or an object
services.nginx.ports contains an invalid type, it should be a number, or an object
services.lnd.ports value ['9735:9735', ':', ':'] has non-unique elements
https://umbrel-paste.vercel.app/7ecbeacb-6bc6-414d-b415-3881f1428f52

lukechilds, 2021-04-29:
It looks like for some reason Umbrel is unable to format your SSD. You can reboot and it will try to format it again.

I'd strongly recommend purchasing the officially recommended parts though: https://github.com/getumbrel/umbrel-os/issues/215#issuecomment-828225379

maravac, 2021-04-29:
OK, but what M.2 SSD enclosure do you recommend if I want to use the M.2 NVMe SSD drive? When I open the link for recommended SSD enclosure, there is only one and is for SATA 2.5" SSD.

Isn't it more about the M.2 NVMe driver or boot driver configuration missing in the umbrel installation? Could this be done, I mean update some config files to mek the NVMe SSD work?

maravac, 2021-04-29:
Just asking, but: could it help if I format that SSD in my PC? I can use the MiniTOol Partition Wizard, whic is capable to format even in ext4.

lukechilds, 2021-04-29:
I'm sorry I'm not sure, we haven't tested any M.2 drives.

maravac, 2021-04-29:
Please, could you discuss it with umbrel developers team? I'm pretty sure that I googled and found some blogs and articles where some people say that they are using the M.2 NVMe together with Raspberry Pi. But also there are some notices about config changes, boot drivers etc. Unfortunately I am not familiar with Linux, SSH and Raspberry (this is my first RPi purchased with intention to use it as umbrel bitcoin node).

AaronDewes, 2021-04-29:
> Please, could you discuss it with umbrel developers team? I'm pretty sure that I googled and found some blogs and articles where some people say that they are using the M.2 NVMe together with [...]

Luke is basically the Umbrel developer team, and if Luke doesn't know, I don't think anybody else will.

maravac, 2021-04-29:
[image]

This is, what I see, when I connect that M.2 drive (in the external USB3 enclosure) to my WIndows 10 PC.

maravac, 2021-04-29:
Please, could you look at this one?
[](https://alexellisuk.medium.com/upgrade-your-raspberry-pi-4-with-a-nvme-boot-drive-d9ab4e8aa3c2)

That guy uses the same M.2 SSD enclosure like I have and he also mentioned teh Samsung 970 EVO in the article. He is running the Raspberry Pi with it. I know, not the Umbrel, but still, hardware is the same..

Also please look this, if this one could be helpful too:
[](https://www.jeffgeerling.com/blog/2021/raspberry-pi-can-boot-nvme-ssds-now)

Let me know, if you will have some advice for me then. Thanks again.

maravac, 2021-04-29:
sorry, but the article on medium should be free to read (I did not pay either :)

AaronDewes, 2021-04-30:
> sorry, but the article on medium should be free to read (I did not pay either :)

[image]

lukechilds, 2021-04-30:
>Unfortunately I am not familiar with Linux, SSH and Raspberry

I'd really strongly suggest using the recommended hardware. This is why we have recommended hardware that we've tested, it's so people can get up and running without needing to have knowledge of these topics.

I'm sure NVMe drives will work with Umbrel but it's not something we've tested and isn't a high priority for us right now so unfortunately not something we can help you with.

maravac, 2021-04-30:
OK, so you rather recommend do use the classic SATA 2.5" SSD in external enclosure (and connected to the USB 3.0 port of Raspberry Pi) instead of M.2 NVMe SSD, which could be much faster, but is problematic to use in terms we discussed here, right?

lukechilds, 2021-04-30:
Yes exactly, I'd recommend you use the exact hardware listed here: https://getumbrel.com/#start

maravac, 2021-05-03:
Hi again!
I have another M.2 NVMe enclosure, which works with my SSD (M.2 NVME Samsung EVO 970 1TB) and also umbrel detected it correctly and started to work. I installed it yesterday afternoon and umbrel connected correctly and started to download and sync bitcoin blocks. It was running about 1 or 2 hours and already synced some 1.06 %, but then something happened and now it looks like this.

[image]

I left it to run during the nigt, but it's still the same after more than 12 hours.
Now I made a restart form the UI of umbrel, it correctly restarted, but the screen is still the same.

I am attaching both logs here, please, would you try to tell me what's wrong and what would you recommend me to do now? 

[umbrel-1620032764400-debug.log](https://github.com/getumbrel/umbrel-os/files/6414346/umbrel-1620032764400-debug.log)
[umbrel-1620032797800-dmesg.log](https://github.com/getumbrel/umbrel-os/files/6414347/umbrel-1620032797800-dmesg.log)

maravac, 2021-05-04:
Please, would you try to look to those log files I have posted in my previous post and try to answer, what can be the problem?
Thank you.

lukechilds, 2021-05-04:
>what would you recommend me to do now?

As mentioned before, I'd really strongly recommend using the officially recommended hardware on our website.

shamblerid, 2021-05-22:
Maybe unrelated but i had issue's with balenaEtcher writing the OS, Switched to Raspberry Pi Imager and had no issues.

maravac, 2021-05-26:
I gave up with the M.2 NVMe SSD drive. I tried 2 different SSDs and 3 different enclosures, not even one worked with umbrel (all were OK with Windows PC although).
Lately I purchased classic SATA SSD (Samsung 970 EVO) and Axagon external enclosure and everything installed correctly and running, more than a week now. (Syncing blockchans took some 2 days).

Sumando10, 2021-06-29:
> Hi, please, would you try to help me fix my problem or atleast find out what could be wrong wint my stuff? I just tried to run my first umbrel node on Raspberry Pi 4B (8GB) + 1TB M.2 NVMe SSD in [...]

I have the same problem that you, using SSD and Raspberry Pi 4 8GB RAM with argon one enclosure m.2

I have spent my money in something that umbrel does not recognize.

Thanks for keep trying. Have you got do working an m.2 with umbrel?

AaronDewes, 2021-06-29:
For the Argon case, you **need** to use the official argon power supply:

https://www.argon40.com/argon-one-power-supply-for-raspberry-pi-4.html
https://www.argon40.com/argon-one-power-supply-for-raspberry-pi-4-eu.html
https://www.argon40.com/argon-one-power-supply-for-raspberry-pi-4-uk.html

maravac, 2021-06-29:
No, M.2 SSD just did not work with any enclosure I tested. I just gave up, purchased classic 2.5" SSD (Samsung SSD 870 EVO), standard USB 3.0 enclosure and installed the Umbrel. It is running without any issues now.

Sumando10, 2021-06-29:
> For the Argon case, you **need** to use the official argon power supply:
> 
> https://www.argon40.com/argon-one-power-supply-for-raspberry-pi-4.html
> https://www.argon40.com/argon-one-power-supply-for-raspberry-pi-4-eu.html
> https://www.argon40.com/argon-one-power-supply-for-raspberry-pi-4-uk.html

Thanks, I will buy it because I can not get working a keyboard on my raspberry pi 4B.

It might be the power supply, I suppose.

@maravac My m.2 is working using the external enclouser, similar at you linked (orico brand), so I guess the problem was the energy supply ,

maravac, 2021-06-29:
I don't think so. I have two different power supply (one of them is original Raspberry) and still had issues with the M.2 NVMe SSD drives. Also I tried two different brands (Samsung and AData) as well as the different M.2 external enclosures. I have got different issues but result was that the Umbrel or even did not start or it did but got stucked during blockchain synchronizing.

Sumando10, 2021-06-29:
> I don't think so. I have two different power supply (one of them is original Raspberry) and still had issues with the M.2 NVMe SSD drives. Also I tried two different brands (Samsung and AData) as [...]

Try to flash the SD memory with umbrel every time you try with other m.2 

In my case, it worked.

rheticus, 2021-07-29:
Hi, I've just installed and set up my umbrel node but I'm unable to connect to the SSD.
The issue seems to be related to /dev/sda and /dev/sda1 in particular.

After getting errors in the debug script (Warning: Error fsyncing/closing /dev/sda: Input/output error.... The file /dev/sda1 does not exist and no size was specified) I tried doing an fsck of /dev/sda1 and got the error message below.

_fsck.ext2: No such file or directory while trying to open /dev/sda1
Possibly non-existent device?_

After a reboot I tried this command again, and now it gives this:

_ext2fs_open2: Bad magic number in super-block
fsck.ext2: Superblock invalid, trying backup blocks...
fsck.ext2: Bad magic number in super-block while trying to open /dev/sda1

The superblock could not be read or does not describe a valid ext2/ext3/ext4
filesystem._ 

At this point I'm completely stumped. Anyone have any ideas for what I can try next?

kevinelliott, 2021-08-19:
No luck for me with ARGON ONE m.2 case at all. Using official power supply from Argon, they even sent me replacement USB connectors. I have tried 4 or 5 different m.2 SSDs. Incredibly frustrating. At this point I have been trying for 4 months. About to give up on ARGON altogether.

manan19, 2021-08-25:
I had a similar issue and the solution @rheticus provided worked for me

> reformatted it as APFS.

bensig, 2021-10-02:
I had an issue with this error on a fresh install of Umbrel using an external drive with USB 3.0

"You have multiple external drives connected to your Raspberry Pi. Please remove the drive(s) you don't want to use for Umbrel"

I ssh'd into my device using: ssh umbrel@umbrel.local

Then I ran the following on my 1TB drive:
`fdisk -l`

Discovered the 1TB drive was /dev/sda - so I partitioned it
`parted /dev/sda`
`mklabel`
* New disk label type? `gpt`                                                  
`mkpart` and use the following options:
* Partition name?  []? `nolabel`                                                  
* File system type?  [ext2]? `ext4`                                      
* Start? `1`                                                                  
* End? `1000GB`
`print`

Next verity that looks correct:
```
Disk Flags: 

Number  Start   End     Size    File system  Name  Flags
 1      1049kB  1000GB  1000GB  ext4
```
Then `quit` parted program by typing `quit`

Next create a filesystem:
`mkfs.ext4 /dev/sda1`

Restart Umbrel and it will find the correct device and drive - works perfectly.

AaronDewes, 2021-10-02:
Thanks for this guide, @bensig. You could change/add `ssh umbrel@umbrel`, without the .local, it is recognized by more operating systems.