getumbrel/umbrel-os #219: How to connect the TREZOR hw wallet?

maravac, 2021-05-26:
Hi, please would you help me to with connecting my TREZOR hw wallet to the umbrel? 

I installed the Spectre Desktop app in umbrel, nut my TREZOR is not recognized when I try to connect it  via USB. My setup is umbrel running on Raspberry Pi and I connect my TREZOR wallet to the USB port of my PC. TREZOR is OK as it works with TREZOR Suite software, but I don't know, how to configure it with umbrel. Mayge some short tutorial or just few steps I need to to do in my umbrel (or Spectre Desktop umbrel app)?

Thanks.

AaronDewes, 2021-05-26:
We use GitHub mainly for development purposes and to report bugs, not to answer questions. To get support with your problem and ensure more people see your question, please post in our forum at community.getumbrel.com/.

I'm not sure how to do this, but I'm sure you'll get an answer on the forum.

lukechilds, 2021-05-31:
@maravac Electrum works well for Trezor and can be connected to Umbrel.

closed 2021-12-16 by jonsyu