getumbrel/umbrel-os #220: Web interface didn't start on first boot

markshep, 2021-05-31:
I ssh'd in and ran the debug script which uploaded this: https://umbrel-paste.vercel.app/1cacc1c2db3b2f08fcab91cfc7779795

It looks like it formatted the SSD, but then umbrel-external-storage.service failed with a dependency issue :-(.

AaronDewes, 2021-06-01:
It seems like your SSD (enclosure) uses too much power or you got a bad power supply.

markshep, 2021-06-01:
I'm using the official Raspberry Pi power supply so it must be the SSD.  I used a Crucial one that I happened to have lying around, but it's an older M500 model rather than the MX500.  It says 1.7A on the back and looking at the [Raspberry Pi Power Supply page](https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md) it says the model 4B can only supports up to 1.2A for peripherals :-(.

Is there actually a bit in the logs somewhere which told you it was a power issue, or did you just know that from having dealt with these issues before?

[image]
[image]

AaronDewes, 2021-06-01:
> Is there actually a bit in the logs somewhere which told you it was a power issue, or did you just know that from having dealt with these issues before?

The logs told me the SSD didn't get connected properly, and having dealt with this issue probably 100s of times tells me that it's a power issue

markshep, 2021-06-01:
Looking around at photos of the MX500 on the internet they also say 1.7A on the bottom, but I guess if you know they work then they must actually draw much less current in day to day usage, so I'll buy one of those.  Thanks for your help!

closed 2021-06-01 by markshep

AaronDewes, 2021-06-01:
> Looking around at photos of the MX500 on the internet they also say 1.7A on the bottom, but I guess if you know they work then they must actually draw much less current in day to day usage, so I'll buy one of those. Thanks for your help!

It often also depends on the SSD enclosure, not only the SSD itself

markshep, 2021-06-03:
I used the recommended enclosure, so it was only the drive that was different.  I've now got an MX500 and installed it and everything is working now:

[image]