getumbrel/umbrel-os #228: Multiple external drives found

giorgioalbertin, 2021-07-03:
During the start of Umbrel OS the following error is reported: 
"Multiple External Drives Found. Umbrel only works with one external drive of atleast 1TB size. Please disconnect any additional external drives and restart your Umbrel."

I installed Umbrel on a USB stick because the Micro SD are usually damaged. Is it possible to allow the use of a second USB device as long as it is lower than the other? For example, Umbrel OS will choose the external disk with bigger capacity as storage.

AaronDewes, 2021-07-03:
> I installed Umbrel on a USB stick because the Micro SD are usually damaged.

You can easily replace/reflash them, because all user data is on the SSD.

> I installed Umbrel on a USB stick because the Micro SD are usually damaged. Is it possible to allow the use of a second USB device as long as it is lower than the other? For example, Umbrel OS will choose the external disk with bigger capacity as storage.

No, that's not possible yet.

micahjcf, 2021-07-22:
Where would this be in the code?

alex27riva, 2022-06-20:
Hi, I have the same issue.
I want to use an USB stick for channel backup

alexander-koval, 2022-08-03:
Hello! I have same issue. I want to use additional storage for backup too.

eisenlohr, 2022-10-08:
It would be really useful to have the option of adding more than one external storage device. Is this a feature that is currently in the works?

ybaidiuk, 2023-10-16:
we really need this feature. Probably on weekend i can try to implement it. but 5% change of success.