getumbrel/umbrel-os #229: Move installation to SSD

diego-G, 2021-07-06:
In order to free the SDCARD and boost the performance, I suggest to create an option to create a partition and move the installation to the SSD.

lukechilds, 2021-07-06:
Thanks for the suggestion, this is actually what we already do by default. See here: https://github.com/getumbrel/umbrel/blob/b2c034c95ed7cbcdc378955cdd0eee63a74eca3d/scripts/umbrel-os/external-storage/mount#L119-L145

closed 2021-07-06 by lukechilds

diego-G, 2021-07-06:
I see that you have a section in the other repo's readme to install the OS in a SSD : https://github.com/getumbrel/umbrel/tree/b2c034c95ed7cbcdc378955cdd0eee63a74eca3d#step-1-download-umbrel

However, I've followed the instructions on this repo which leads me to a different situation: SD Card is stuck with Umbrel OS installed. The Raspberry doesn't turn on without the card. How would you suggest me to proceed @lukechilds ?

lukechilds, 2021-07-06:
That's expected, the SD card is required to boot, but Umbrel and Docker run off the SSD.

diego-G, 2021-07-06:
That was the option that I was suggesting: moving the boot partition to the SSD in order to freed the SD card.

lukechilds, 2021-07-06:
Ahh I see. We like to run the OS on the SD card so for recovery users can easily re-flash the SD card and get a working system again.

While we could make that logic work with a full OS on the SSD, it would take quite a bit of work and wouldn't offer much advantage over the current solution of running the OS on the SD card and just bind mounting the IO heavy parts of the OS onto the SSD.

At some point in the future we may support running Umbrel OS entirely on external storage but it's not high priority for us to implement right now.