getumbrel/umbrel-os #231: SSD cannot be found

ando999, 2021-07-28:
Most of the time, after upgrading and re-updating my alias for LND, I have to disable the UAS protocol so Umbrel can see my SSD, ....I use the following command.

`sed -i 's/  blacklist_uas/  #blacklist_uas/g' /home/umbrel/umbrel/scripts/umbrel-os/external-storage/mount
`

I was given this command by one of the guys that was working on the project previously.  Is it possible to fix this so I don't have to run this commend and then reboot afterwards?

fultonm, 2021-10-30:
Ugh. I was trying to figure out why my SSD wasn't loading, and why it did before. Before, I manually mounted my SSD, which bypassed all the umbrel OS scripts. Then, I decided to start fresh and let Umbrel OS manage my storage. I had no idea why I was getting the following error when it worked perfectly before:

It seems that this blacklist function is not ready for primetime??

lukechilds, 2021-10-31:
Are you using the official Raspberry Pi power supply and the reccomended SSD and enclosure on our website?

The Pi has underpowered USB ports and can be very unreliable with certain combinations. Blacklisting the UAS drivers offers significantly improved stability but only under most hardware we've tested. But some combinations of hardware can still have issues.

fultonm, 2021-10-31:
No I am using the 5.1v 3.5a power supply that came with the Canakit Raspberry Pi 4 set up kit.
The hard drive is Crucial X6 1TB Portable SSD - Up to 540 MB/s - USB 3.2 - External Solid State Drive, USB-C - CT1000X6SSD9
I think this combination works fine, in fact I've been hammering it with experiments to debug https://github.com/getumbrel/umbrel/issues/1077 and it's been rock solid
The black list script was kind of like a hidden gotcha.
How would you feel about prompting the user at start up that their configuration is unsupported and to proceed with caution?

lukechilds, 2021-11-01:
We've had a huge amount of reports from users experiencing problems with the Canakit power supplies. Would strongly advise changing to the official Raspberry Pi power supply.

>How would you feel about prompting the user at start up that their configuration is unsupported and to proceed with caution?

It's a nice idea although I think it could cause some confusion for non-technical users, when there hardware config may work well even if it's non-standard. It would also mean maintaining and constantly updating a list of supported/unsupported hardware which was the problem we had before when we were manually disabling UAS for certain devices and then decided to stop doing that and just blanket disable for all devices.

masteruan, 2022-11-27:
Same issue: hard drive Crucial X6 1TB Portable SSD - Up to 540 MB/s - USB 3.2 - External Solid State Drive, USB-C - CT1000X6SSD9
I've changed my power supply...and that's it! Awesome! All works very nice