getumbrel/umbrel-os #237: Lightening payment confirmation error

trentycents, 2021-08-19:
When making payments / paying an invoice using my my lightening wallet I get the attached response when my payment goes through successfully. At first I thought the payment had failed, but then realised its the confirmation message or screen that fails to load.

Code that displays is:

[image]

<html> <head><title>502 Bad Gateway</title></head> <body> <center><h1>502 Bad Gateway</h1></center> <hr><center>nginx/1.17.8</center> </body> </html>

trentycents, 2021-08-19:
Umbrel Version 0.4.0

jonsyu, 2021-12-17:
Closing due to inactivity, but if this issue still remains, please post about it on our community forum (https://community.getumbrel.com)

closed 2021-12-17 by jonsyu