getumbrel/umbrel-os #248: [API - REST] GET Address UTXO missing

zatJUM, 2022-01-20:
Hi

Try this:
```
curl -sSL https://mempool.space/api/address/3Adsw48SVXTGU5LNhXwnLReAE6J81yENZk/utxo | jq length
1
```
Then, on your umbrel node:
```
curl -sSL http://127.0.0.1:3006/api/address/3Adsw48SVXTGU5LNhXwnLReAE6J81yENZk/utxo | jq length
parse error: Invalid numeric literal at line 1, column 10
```
Tested on 0.4.11 version

louneskmt, 2022-01-23:
@wiz

softsimon, 2022-01-23:
This is an `Blockstream/esplora` specific API endpoint that we have not re-implemented in our NodeJS/bitcoind backend since we are not using it.

If you open an issue or make a PR on https://github.com/mempool/mempool/issues I guess there is possible to re-implement this API for the Umbrel setup with `romanz/electrs`.