getumbrel/umbrel-os #249: Can't connect via SSH to umbrel node

tifDev, 2022-02-02:
Via web I access with no issues but via SSH I can't :/

Here's what I'm doing:

ssh umbrel@umbrel.local
Password: Same as using in web access.

I get: **Permission denied, please try again.**

louneskmt, 2022-02-10:
Have you tried to restart Umbrel and login back into the WebUI, then try again?

Veeeetzzzz, 2022-03-24:
I've tested this with a fresh install and after the password is updated via umbrel.local I can login in via SSH with the new password. Sure you're not mistyping the password in SSH? It can be difficult as you don't see the text being entered.

What OS are you using? Have there been any changes to your SSH keyring? If you're on Windows - navigate to C:\Users\[user]\.ssh and check the known_hosts file to see if there's an entry for your Pi device. 

If there's an entry it means your password is definitely being entered incorrectly. 

If you're still having issues I would post some further details - OS in use & try a restart/reset of your device.

jeremycadams, 2022-07-22:
On a fresh install, yes, SSH should and has worked fine for me. However, after running for roughly 8 months with no hiccups, I had a networking meltdown (nothing to do with my Umbrel on Raspberry Pi 4), and had to entirely rebuild my network from scratch. The Umbrel had to get the Ethernet moved from behind my normal router, to the cable modem/router, back to my internal router (Ubiquiti Edge X-SFP) and throughout all that time it kept running and working with LN, Bitcoin network, Tor, etc. However, now my SSH port (22) on my Umbrel is closed for some unknown reason. I've scanned it with nmap:

<pre>nmap -p22,113,139 192.168.1.39
Starting Nmap 7.80 ( https://nmap.org ) at 2022-07-22 18:02 EDT
Nmap scan report for 192.168.1.39
Host is up (0.0052s latency).

PORT    STATE  SERVICE
22/tcp  closed ssh
113/tcp closed ident
139/tcp closed netbios-ssn

Nmap done: 1 IP address (1 host up) scanned in 0.08 seconds
</pre>

and also restarted my Umbrel and checked logs (yes, the Umbrel is at 192.168.1.39 on my internal network).

My Umbrel hardware (except for the ethernet cord) hasn't been unplugged, damaged, or changed in any way. My guess is that either Pi 4's have a cheap ethernet card or chip which the OS doesn't know how to deal with when new DHCP-provisioned IP addresses show up, so it shuts off SSH, OR the SSH software is just not working correctly within Umbrel's Dockerized container approach to things, and gets shutdown somehow and can't be restarted from the web dashboard. (By the way, during this entire time I have no problem using the web interfaces to all of the other software on my Umbrel.)

friedger, 2023-11-15:
For me ssh port is open. Web access: no problem. ssh and login via terminal on node: permission denied and login incorrect