getumbrel/umbrel-os #265: Question | MicroSD wearing down

gustavobrossi, 2022-11-01:
Hi,
Recently I've started getting alerts over the UI from my node saying to replace my microSD card as it could fail soon.
I am curious how you are able to identify that - I've tried to dig the repositories but couldn't find anything, would you please point me to the place you are managing this alert?

Thank you!

PlebeiusGaragicus, 2022-11-12:
I'm wondering the same and also read that the latest Umbrel update disabled this feature.  I'd love some kind of explanation.

Also, have we fixed the issue where Umbrel won't load on a USB thumb drive?  I have several high-quality USB sticks that are likely to last longer and won't run into this issue.

Also again, can we make the OS load from the USB stick into RAM so that it doesn't wear it out?  Something like the "Overlay File System" setting rpi had?

justingoldberg, 2023-05-05:
To extend the life of an ssd, you should only partition up to 70% of the ssd. IE: The less you use, the longer it will last.