getumbrel/umbrel-os #269: No apps on app store

MateoRbt, 2022-11-22:
Fresh install of umbrel-os v0.5.3  doesn't show apps and can't find them on search bar

IMPranshu, 2023-07-08:
Are you facing this issue? Can you describe the issue and attach screenshots

mroxso, 2023-12-14:
I'm facing the same issue.
[image]

mroxso, 2023-12-21:
My problem go solved by executing `sudo rm -f ~/umbrel/repos/https---github-com-getumbrel-umbrel-apps-git/.git/index.lock && sudo rm -f ~/umbrel/repos/https---github-com-getumbrel-umbrel-apps-git/.git/ORIG_HEAD.lock && sudo ~/umbrel/scripts/repo update`
I found this here:
https://community.umbrel.com/t/umbrel-troubleshooting-guide/3632
(Cannot find an app in the Umbrel App Store)