getumbrel/umbrel-os #270: BTCPay Server installation stuck in a loop after uninstallation

SovNym, 2023-03-03:
## 🚨 BTCPay Server installation stuck in a loop after uninstallation

### Current behavior
👉 After BTCPay Server was installed and later deleted, attempting to reinstall it leads to a situation where the installation process appears to be stuck and continuously attempts to install the app, without making any progress.

### Expected behavior
👉 The BTCPay Server app should install successfully on the Umbrel node without getting stuck in a loop.

### Steps to reproduce
1. Install BTCPay Server on Umbrel node
2. Delete BTCPay Server
3. Attempt to reinstall BTCPay Server on the same Umbrel node

### Additional information
- Umbrel version: 0.5.3
- BTCPay Server version: Version 1.7.12
- Any error messages encountered: None on gui

📷 Please find attached a screenshot of the installation screen.

[image]

IMPranshu, 2023-07-08:
Try restarting your Umbrel once.

ybaidiuk, 2023-07-25:
have a same.  did remove docker container.