getumbrel/umbrel-os #273: Missing Documentation - 2FA Lockout

konjoinfinity, 2023-06-03:
This should be in the GitHub documentation, when users lock themselves out of their Umbrel.

1) Download and install putty - https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html
2) Open putty and type in umbrel.local for the hostname.
3) Click Open or hit Enter. 
4) Then type umbrel for the username.
5) Then type your password, it's same password you use to log in to the Umbrel dashboard.
***If you haven't yet run through the setup process, the password will be set to `moneyprintergobrrr`.
6) Then type nano umbrel/db/user.json - This will open the json file for editing.
7) Using the down arrow, scroll down until you see " "otpUri": "otpauth://totp/umbrel?secret=blblablabla&period=30&digits=6&algorithm=SHA1&..." - or similar.
8) Delete the line so it looks like “otpUri”: “”.
9) Then hit Ctrl-O then Enter to save the file and lastly Ctrl-X to close the file.
10) Type logout and hit enter to close Putty.
11) Login to your umbrel normally, be sure to re-enable 2FA after regaining access.

References:

- https://community.getumbrel.com/t/2fa-recovery-lost-phone-no-code/5135/36?u=wesmsl
- https://github.com/getumbrel/umbrel-os

IMPranshu, 2023-07-08:
I think this is more of a workaround/hack and including it in the official docs would mean the security of 2FA is compromised and they know it. 

Anyways this is just a backdoor nothing else they might be removing this in the future realise.

smolgrrr, 2023-08-08:
It's also on the Official Troubleshooting guide https://community.umbrel.com/t/official-umbrel-troubleshooting-guide-and-faq/9873