getumbrel/umbrel-os #279: Theoretical question

ybaidiuk, 2023-09-13:
I saw this tweet, and it is impressive, but I have this question:
imagine that i opened a channel for, 1 btc and my node broke.
and I send 0.8 btc through the forwarding process.
and my last backup file has info that I still have 0.5 btc on my side.
I restore the node, another node thinks that I want to deceive her and takes all the funds from the channel.
So I will lose, 0.2 btc?

lukechilds, 2023-09-15:
No, we don't restore Lightning channel state because that would be risky for the reasons you mentioned. We restore from LNDs static channel backup. This doesn't update per state change, it only updates per channel open. As long as you have a backup since the last channel you opened, the recovery process can be initiated. It's not trustless, it requires your peers to cooperate to successfully recover your funds, but there are some incentives at play to encourage them to work with you. It also results in force closing all channels to re-claim the funds on-chain.

ACINQ use a similar model and wrote a good article on it: https://medium.com/@ACINQ/adding-data-loss-protection-to-eclair-598c62494096

closed 2023-09-15 by lukechilds

ybaidiuk, 2023-09-15:
Thanks for explaining,
What If peer don't want cooperate, or give me wrong data that's he can give me penalty later?